teslamotors/vehicle-command #114: Feature Request: Summon

stx, 2023-12-23:
We have a Summon streaming API implementation which many customers have enjoyed. The new third party apps API authentication doesn't support streaming API authentication, so users lose the ability to use Summon.

It would be great to add Summon to the new vehicle command protocol so it covers the full range of vehicle controls.

Mushy-Snugglebites-badonkadonk, 2024-01-05:
Summon with the Apple Watch has always been clutch. Sad to see this functionality go. Would be amazing to have it back again and before the next holiday update.

jandieg, 2024-01-10:
Please restore this

shh0558, 2024-01-16:
It is very unfortunate that Smart Summon cannot be run on other apps. I hope it returns to normal as soon as possible.

GuruDanny, 2024-01-21:
Tesla's own summon feature is not so finished - Tessie had refined that functionality and it should be available to Tesla owners and NOT restricted by Tesla Corp.

smathias503, 2024-04-21:
Bring it back to Tessie so we can do it from our watches.  All the stooopid people who shouldn’t have FSD since they’re dumb are ruining it for the rest of us, causing Tesla to restrict implementation & more aggressive development.  I think those drivers should’ve 
_never_traded in their Volvos’!

thefireblade, 2025-01-22:
We currently do not have a roadmap for this unfortunately, will keep this open as a feature request for now

Holki-cozy, 2025-03-22:
Please restore this

michaelcbarr, 2025-11-24:
Please add this functionality back !

Mushy-Snugglebites-badonkadonk, 2025-11-24:
> Please restore this

> Please add this functionality back !

This isn’t a Tessie problem. This isn’t a Tesla problem (but it is, more in a sec), this is a NHTSA problem. e.g., rules & regulations. Back to Tesla, because Tesla has now created an official API and has forced all developers to onboard, the “hacked” (reverse engineered) API that developers were using has been shutdown. Tessie no longer has access to Summon since Summon is not part of the official API. No developer has access to my knowledge unless there is another “work around” found. 

https://developer.tesla.com/docs/fleet-api/announcements#2024-03-26-shutting-down-legacy-vehicle-api-endpoints