teslamotors/vehicle-command #115: Feature Request: Battery Preconditioning

stx, 2023-12-23:
We've regularly heard from customers over the years that they'd like to precondition their battery for fast charging without navigating to a Supercharger.

It would be great to have a set_battery_preconditioning command, perhaps with "level": "standard" and "level": "fast" parameters.

ckeylan, 2024-08-25:
Tesla does not show all dc chargers in Türkiye and it would be nice to precondition the battery manually or at least when we choose a destination from the map to go, there can be an option to say this is a dc fast charger so precondition the battery when car is approaching to the destination.

crossan007, 2025-01-04:
I'd really like to see this implemented. 

Also, is there any documentation about how to invoke the battery precondition command locally from within the vehicle? 

If S3XY buttons can do it, then there should be a developer friendly way, right?

thefireblade, 2025-01-22:
I've filed an internal ticket for this feature request, no ETA though

alexw23, 2025-05-28:
@thefireblade I notice the car will invoke preconditioning when you Navigate to an SC in the app. Not sure when this update was released but it never used to be like that. 

Would be great to have [navigation_sc_request](https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#navigation-sc-request) working, or some documentation around how it works. I've tried a LocationGUID and an internal ID (from sniffing the app traffic), doesn't matter what ID I use, I can't get the car to respond.