teslamotors/vehicle-command #117: Elliptic ScalarMult function availability in other languages

joshendy, 2023-12-23:
Sorry, this isn't exactly on topic - I'm just desperate. I've been trying to convert this library to Java so I can natively run this on Android. I'm stuck on the elliptic.P256().ScalarMult function. I can't find an equivalent elliptic P256 scalar multiply function in bouncy castle or any other crypto library. Any help or direction to go looking in would be greatly appreciated.

RonnyWinkler, 2023-12-23:
Hi @joshendy 
Could you perhaps share your Java version? I try to implement the signed command in JJ (NodeJS) and can't find what's really needed to build the correct base64 body. Many thanks :-)

joshendy, 2023-12-23:
Whoops. Bouncy castle's default ECMultiplier works fine as a replacement.

closed 2023-12-23 by joshendy