teslamotors/vehicle-command #120: getting error while executing ./tesla-control lock command

NikhilTank35, 2023-12-23:
I am getting the below error while executing ./tesla-control lock command. Could you please help me with this error?

```
~~/vehicle-command-main/cmd/tesla-control/bin$ ./tesla-control lock
Error loading credentials: could not load token: The specified item could not be found in the keyring
```

fabianhu, 2023-12-23:
try to provide the required keys and stuff this way, so you don't rely on the key management of your operating system: 
`./tesla-control -key-file {keyfile} -token-file {tokenfile} -vin {vin} {command_string}'`

NikhilTank35, 2023-12-24:
Thanks for your reply.

I tried providing the required keys like the above but I am now getting below another error.

```
Downloads/vehicle-command-main/cmd/tesla-control/bin$ ./tesla-control -key-file ../../tesla-keygen/bin/public_key.pem -token-file outh_token_file.txt -vin vin_dummy_no lock
Error loading credentials: asn1: structure error: tags don't match (2 vs {class:0 tag:16 length:19 isCompound:true}) {optional:false explicit:false application:false private:false defaultValue:<nil> tag:<nil> stringType:0 timeType:0 set:false omitEmpty:false} int @2
```

sethterashima, 2024-01-10:
It looks like you're providing a public key to the `-key-file` argument. The private key is used to sign the command.

That being said, we should provide a more helpful error message here.

thefireblade, 2025-01-22:
Closing old issue

closed 2025-01-22 by thefireblade