teslamotors/vehicle-command #123: user_data vehicle_cmds vehicle_charging_cmd scopes are not applying in https://auth.tesla.com/oauth2/v3/token URL

NikhilTank35, 2023-12-28:
Hi Tesla developer team,

We got the client id and client secret and selected all scopes like the one below.

[image]

then user goes into 
`https://auth.tesla.com/oauth2/v3/authorize?client_id={client_id}&redirect_uri=http://localhost/tesla&response_type=code&scope=openid%20offline_access%20user_data%20vehicle_device_data%20vehicle_cmds%20vehicle_charging_cmd&state=db4af3f87`
and enter credentials.
after, we got the code in the redirect URL, and using that code, we got an access token like below.

[image]

when we decode this access token, we found that it only contains openid, offline_access, and  vehicle_device_data scopes. it doesn't have user_data vehicle_cmds vehicle_charging_cmd scopes. Could you please help me with this?

[image]

NikhilTank35, 2024-01-01:
@supleshg Thanks, 

Do you know how to revoke the access token? eg. need to follow some steps or does any endpoint need to hit?

andig, 2024-01-25:
Same problem here. Issued tokens do not contain  all scopes from app.

> Here's what helped: I revoked the previously issued access token for my account. Once you revoke the previous token, the next attempt to get an access token should prompt you to select the checkbox with the required scope. Hopefully, this solution works for you.

@supleshg do you mean to revoke a single access token and then create a new one for the same app? I'm confused why the user should then choose the scopes? They are already determined by the code request and must not be modified?

andig, 2024-01-26:
Update: I realise that the token auth process does not make clear which scopes have been requested. That's a showstopper as it is unclear for the user which scopes have even be requested. See https://github.com/teslamotors/vehicle-command/issues/63.

h4ckdepy, 2024-10-23:
> @NikhilTank35 Click on this [link](https://accounts.tesla.com/en_US/account-settings/security?tab=tpty-apps) then in `Third Party Apps section` find button for `manage` next to your application name. There you can `revoke` the access. Alternatively, you can loging to Tesla website, go to your profile settings then manage third party apps.

How can developers of  **.cn** revoke third-party application authentication? No matter how much authorization is given to the user_data scope now, it cannot be added

thefireblade, 2025-01-23:
Hey @NikhilTank35, I know this is late, I've redacted parts of your issue body that includes sensitive information. Moving forward such a request can be forwarded to https://fleetapi-support.tesla.com/

closed 2025-01-23 by thefireblade