teslamotors/vehicle-command #126: getting "Unable to Share Vehicle" error after opening QR code link

NikhilTank35, 2023-12-29:
it is showing the below error after opening QR code link 

[image]

Unable to Share Vehicle
Adding a virtual key for "blinded- standard.000webhostapp.com" is forbidden because you have not granted "blinded-standard.000webhostapp.com" access to your account

@ThomasAlxDmy @sethterashima @fabianhu Can you please help us with this?

fanhaipeng0403, 2023-12-29:
I am also facing a similar problem. Can someone help me take a look? Ahhh.
[image]

fanhaipeng0403, 2023-12-30:
@NikhilTank35  Hello, any progress with this question?

mmrazik, 2024-01-08:
I have the same issue. Can't even add existing apps like tessie.com. I am in Europe.

sethterashima, 2024-01-10:
The user needs to approve the app using the OAuth flow before the app will allow them to add the virtual key to their vehicle (https://developer.tesla.com/docs/fleet-api#authentication).

When logged in using the mobile app user's account (as opposed to the developer's account), the app must appear in https://accounts.tesla.com/en_us/account-settings/security?tab=tpty-apps.

sethterashima, 2024-01-10:
Closing this ticket because it appears to be working as intended. If the problem persists even if the user has already approved app access through the OAuth flow, then create a new Issue.

mmrazik, 2024-01-11:
@sethterashima  please reopen it. The problem is that the screenshot are from the app after accessing the tesla.com/_ak/example.com. What other way there is to approve the app? 
My flow is the following:
1. Go to https://tesla.com/_ak/tessie.com
2. Open it in the mobile -- the Tesla App Opens
3. The screenshots appear

Expected result:
3. Instead of the screenshot I get a promopt to approve or deny the Oauth

mmrazik, 2024-01-11:
Nevermind. I'll create a new issue.

mmrazik, 2024-01-11:
Created https://github.com/teslamotors/vehicle-command/issues/137 but it really is the same issue.

closed 2025-01-22 by thefireblade