teslamotors/vehicle-command #127: Http Proxy - Production Documentation?

eclipsed4utoo, 2023-12-30:
The documentation for the Http Proxy works fine for local development, but there isn't much information for how to setup the http proxy for production use.

I have a mobile app that's written in a language that doesn't allow me to use the SDK directly, so it would seem I need to use the http proxy.

I have my mobile app working perfectly with the Fleet API. I have my mobile app working perfectly with the http proxy running locally on my Mac. Now I'm stuck at what to do next. Obviously I can't continue to run the http proxy locally on my mac, so I need to run it somewhere public, correct?

I have my public key hosted at my domain..

https://mycooldomain.com/.well-known/appspecific/public-key.pem

Do I need to run a VPS for the http proxy? 
- If so, does it need to use my domain? or can my app call the IP address of the VPS directly?

What key/certs do I use to run with the http proxy? Locally, we just create unencrypted key/cert for dev purposes, so I'm assuming I should not do that for public/production.
- Do I use my SSL key/cert for my domain? 
- Do I buy another SSL key/cert and use that?

Trying to figure out how to get my app to support the http proxy in a production environment since the expectation that the Fleet API use for commands will be depreciated very soon(within the next week).

psmedley, 2024-01-27:
I installed a nginx proxy that handles the SSL key/cert for the domain, and passes the query to tesla-http-proxy

eclipsed4utoo, 2024-01-27:
> I installed a nginx proxy that handles the SSL key/cert for the domain, and passes the query to tesla-http-proxy

Thanks for the reply. Are either of these tools sitting on the server where the domain is pointed? Or are they on a different server/machine?

My domain is hosted by namecheap, so I'm trying to figure out if I need to host it myself on my own server/machine for this to work.

psmedley, 2024-01-27:
In my case, nginx and tesla-http-proxy are running on the same hardware. tesla-http-prooxy  only listens on localhost -  so they *have* to run on the same box

iggna, 2024-03-27:
following @eclipsed4utoo I installed the tesla sdk in a heroku instance, but couldn't made it work yet. Created openSSL pems for my custom domain, it is initializing and everything but still getting a 

`2024/03/27 18:35:54 http: TLS handshake error from 172.18.42.149:12210: EOF`

In my logs, wonder if anybody had the same issue or any way to make it work, would appreciate some help!

thefireblade, 2025-01-22:
We have improved the documentation significantly since then, closing this old issue

closed 2025-01-22 by thefireblade