teslamotors/vehicle-command #129: iOS adding public key in app only works if app is closed

simon-norman, 2024-01-02:
To reproduce:

1. Select the https://tesla.com/_ak/your-domain.com link

itsMeDavidV, 2024-01-02:
@sethterashima it would also be nice if we could submit a callback url for this task specifically so that the user can be taken back to the app/web client upon result

sethterashima, 2024-01-02:
Thanks for the report. The mobile app team has reproduced the issue and is working on a fix.

I'll pass along the callback url suggestion. We'd likely require the callback url to be on the same domain.

simon-norman, 2024-01-03:
Thank you! @sethterashima this has been raised elsewhere but even more useful than the callback url IMO would be the ability to pass in a VIN to pre-select the vehicle being authorised, so we can take the user through a flow where they approve their vehicles one by one.

simon-norman, 2024-01-08:
@sethterashima just to check if any update on this?

sethterashima, 2024-01-08:
@simon-norman this was a website (not mobile app) bug that should be fixed. Are you still encountering the issue?

simon-norman, 2024-01-09:
@sethterashima different issue now - now when you select the link, it goes to the splash page where you can select Finish Setup, then when you select that nothing happens

simon-norman, 2024-01-09:
It just refreshes that splash page

itsMeDavidV, 2024-01-18:
> @sethterashima different issue now - now when you select the link, it goes to the splash page where you can select Finish Setup, then when you select that nothing happens

@sethterashima this is still a big issue for many users.

rawmean, 2024-01-26:
One workaround is to scroll up and  tap on "open in app" button when you get the splash screen.
Restarting the iPhone is another workaround.
Both cause bad user experience. I hope Tesla fixes this.

kevin39, 2024-02-01:
Workaround : This works fine when Safari is set as the iOS default browser.
If Chrome is the default browser, this will not work.

p-j-l, 2024-06-28:
We (ev.energy) are also running into this issue when onboarding Tesla vehicles.

thefireblade, 2025-01-22:
Many of these issues would be resolved with https://github.com/teslamotors/vehicle-command/pull/269. Let's open a new issue if this is still persisting

closed 2025-01-22 by thefireblade