teslamotors/vehicle-command #130: Error "invalid_request"

onlyusedtesla, 2024-01-03:
We are trying to set up Telsa as an Auth provider in Auth0. This is to allow logins with Tesla accounts into our web application. We then intend to make use of the generated token to query the Tesla API's. We have configured everything on the Auth0 side, there isn't much to configure just provide the auth and token endpoint, scopes(openid offline_access user_data vehicle_device_data) and client ID and Secret. The final part of the setup requires creating a user account in Auth0 during their first login. For that, we are using the "Fetch User Profile Script", which calls the API to get their user info from Tesla, however we are seemingly unable to make the call due to a region error, we have tested with EU/China and receive the same. please help. 
[image]

eclipsed4utoo, 2024-01-03:
Did you call the `/api/1/register` endpoint for your domain? This is required.

https://developer.tesla.com/docs/fleet-api?shell#setup
[image]

I just used Postman, since this only needs to be done once. 

[image]

I also used Postman to get the partner authentication token, which you get from making this request to `https://auth.tesla.com/oauth2/v3/token` (it's required to make the "register" API request)

[image]

onlyusedtesla, 2024-01-04:
Thank you for the suggestion. We tried generating the partner token, and triple checked the client/secret ID combination, and we cannot get past this error.

[image]
cret ID combination, and we cannot get past this error.

thefireblade, 2025-01-22:
This issue should be redirected to Fleet API Support, closing this old issue.

closed 2025-01-22 by thefireblade