teslamotors/vehicle-command #131: Region routing broken for forwarded requests and cars that don't support Vehicle Command Protocol

stx, 2024-01-03:
If you send a Vehicle Command Protocol request to an access token with "ou_code": "EU" but with a user region of NA, the proxy sends the command to EU, receives a 421, then reroutes to NA, then learns the car doesn't support Vehicle Command Protocol, then forwards the request on to the (wrong) EU region and gives up. Net result: all commands fail on the vehicle.

The root issue is that forwarded requests do not automatically retry the correct region.

```
2024-01-03T14:53:16Z [info ] Received POST request for /api/1/vehicles/VIN/command/door_lock
2024-01-03T14:53:16Z [debug] Executing door_lock on VIN
2024-01-03T14:53:16Z [info ] Starting dispatcher service...
2024-01-03T14:53:16Z [info ] Requesting session info from DOMAIN_INFOTAINMENT
2024-01-03T14:53:16Z [info ] Requesting session info from DOMAIN_VEHICLE_SECURITY
2024-01-03T14:53:16Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/VIN/signed_command: {"routable_message":"..."}
2024-01-03T14:53:16Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/VIN/signed_command: {"routable_message":"..."}
2024-01-03T14:53:16Z [debug] Server returned 421: Misdirected Request: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.na.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2024-01-03T14:53:16Z [debug] Server returned 421: Misdirected Request: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.na.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2024-01-03T14:53:16Z [debug] Received HTTP Status 421. Updating server URL.
2024-01-03T14:53:16Z [debug] Received HTTP Status 421. Updating server URL.
2024-01-03T14:53:16Z [debug] [20e25b4af188a5043a55432e99ba4c52] Retrying transmission after error: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.na.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2024-01-03T14:53:16Z [debug] [f85719b1f9a8ac1135d89aea21861d40] Retrying transmission after error: {"response":null,"error":"user out of region, use base URL: https://fleet-api.prd.na.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements","error_description":""}
2024-01-03T14:53:17Z [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/VIN/signed_command: {"routable_message":"..."}
2024-01-03T14:53:17Z [debug] Sending request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/VIN/signed_command: {"routable_message":"..."}
2024-01-03T14:53:18Z [debug] Server returned 422: Unprocessable Entity: {"response":null,"error":"vehicle does not support the Tesla Vehicle Command Protocol, please refer to the documentation here: https://developer.tesla.com/docs/fleet-api#2023-10-09-rest-api-vehicle-commands-endpoint-deprecation-warning","error_description":""}
2024-01-03T14:53:18Z [warn ] [f85719b1f9a8ac1135d89aea21861d40] Terminal transmission error: vehicle does not support protocol -- use REST API
2024-01-03T14:53:18Z [debug] Forwarding request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/VIN/command/door_lock
2024-01-03T14:53:18Z [debug] [20e25b4af188a5043a55432e99ba4c52] Retrying transmission after error: Post "https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/VIN/signed_command": context canceled
```

@sethterashima FYI since you were working on this.

rawmean, 2024-01-07:
Is token with ou_code = EU and user region = NA a valid configuration? In what situation this is expected? Isn't it safe to just use the `region` and ignore the `ou_code` in the token?

stx, 2024-01-07:
This SDK doesn't appear to use region at all. It relies on ou_codes and redirects (or lack of, in this case) so it is broken for this Tesla account configuration. I suspect it relates to importing cars from NA.

rawmean, 2024-01-08:
I think I encountered another account like this. My workaround was to toggle between NA and EU when it fails. It's ugly, but I hope Tesla fixes it.

andig, 2024-01-25:
We're seing HTTP 421 with [evcc.io](https://evcc.io). Setup is EU developer account with self-hosted users all over the world. We're so far using the EU api endpoint. Due to lack of https://github.com/teslamotors/vehicle-command/issues/29 we're handcrafting the requests (no commands, just reading).

It's unclear if EU api for non-EU vehicle  is a valid scenario?