teslamotors/vehicle-command #132: Changing password invalidates OAuth tokens

cieniewski, 2024-01-04:
Hello, after requesting authorisation from customer and generating third-party tokens, customer changing their password results in tokens stopping working.

Here is the response from `https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/users/me` (access token used):
Code 401
```
{
    "error": "token expired"
}
```

And this is the response from `https://auth.tesla.com/oauth2/v3/token` (refresh token used):
Code 401
```
{
    "error": "login_required",
    "error_description": "Login required",
    "error_uri": "https://auth.tesla.com/error/reference/bd888409-f6e9-4a26-bfc1-a2145c65a785-1704374375976"
}
```

Is this expected behaviour? Is there a way to keep the access after password is changed without asking user to reauthenticate?

itsMeDavidV, 2024-01-05:
This is expected behavior. All active tokens are invalidated upon password change.

sethterashima, 2024-01-10:
Working as intended. Changing a compromised password should invalidate logins.

closed 2024-01-12 by sethterashima