teslamotors/vehicle-command #133: Bug Fixes

Urkman, 2024-01-05:
Hello,

is nobody interested in fixing all the bugs? 
 - Boombox not working
 - Vent windows not working
 - Trunk not working correctly 
 - "users/me" endpoint not working
 - no docs for using the "signed_command" endpoint
 - and more

There are so many open points, but nothing happens :(

Urkman, 2024-01-09:
Seat Heater is also not working using the Proxy...

Urkman, 2024-01-09:
And it is not even documented correctly:
https://developer.tesla.com/docs/fleet-api#remote_seat_heater_request

francinv, 2024-01-26:
> Seat Heater is also not working using the Proxy...

It works, but the documentation does not tell you what the exepcted body is. After some testing this worked for me:

```json
{
    "seat_position": integer,
    "level": integer
}
```

Seat positions:
- 0: Front Left
- 1: Front Right
- 2: Rear Left
- 4: Rear Center
- 5: Rear Right

Level:
- 0
- 1
- 2
- 3

Urkman, 2024-01-27:
Thanks... Would be great to have some documentation ;)

thefireblade, 2025-01-22:
Closing old issue, documentation has been improved significantly since https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-commands#remote-seat-heater-request

closed 2025-01-22 by thefireblade