teslamotors/vehicle-command #134: Error: Client / Secret ID combination Error. Please help. Thank you.

onlyusedtesla, 2024-01-05:
We tried generating the partner token, and triple checked the client/secret ID combination, and we cannot get past this error.

{
    "error": "unauthorized_client",
    "error_description": "We don't recognize this client_id and client_secret combination",
    "error_uri": "https://auth.tesla.com/error/reference/96879f65-4f18-4f1c-ab97-4bce6265cc8a-1704326384901"
}
[image]

stephanmo, 2024-01-08:
The problem I also face is, that my application registration(s) have vanished in my developer account.
Also creating a new application fails.
When trying to create a new application, i get a success message, but i cannot find the application anywhere respectively i do not get redirected after successful creation.

techguy993, 2024-01-13:
Yes, I am having the same problem, though my Tesla developer account shows my client_id and client_secret, so not sure what is wrong.

onlyusedtesla, 2024-01-13:
I got new numbers / client / id / secret and same error.

jdolderer, 2024-01-15:
Same issue here. Any help welcome!

onlyusedtesla, 2024-01-17:
even our new numbers didn't work. Can someone please help here?

hauserkristof, 2024-01-19:
I have the same problem, with a newly created app. Fortunately old ones work well.

onlyusedtesla, 2024-01-19:
I wish someone at Tesla would get back to us / fix this or turn on billing so we can build our application. This is blocking us. Please tesla engineering help us. Thank you

techguy993, 2024-01-21:
Tesla Engineering?  Any Help Please?????

gilhrpenner, 2024-01-22:
Also having the same issue, posting here for viz and subscribing.

onlyusedtesla, 2024-01-22:
Tesla engineering please get back to us about this. Everyone is blocked. Thank you. Its been 2 weeks now.

ThomasAlxDmy, 2024-01-24:
Hey all - have you tried setting `Content-Type: application/x-www-form-urlencoded` ?

gilhrpenner, 2024-01-24:
> Hey all - have you tried setting the header `Content-Type: application/x-www-form-urlencoded` ?

Yep, below is the curl from my postman, but have also tried with the curl provided in the docs.

```
curl --request POST \
  --url 'https://auth.tesla.com/oauth2/v3/token?grant_type=client_credentials&client_id=[client_id]&client_secret=[client_secret]&scope=openid%20offline_access%20user_data%20vehicle_device_data%20vehicle_cmds&audience=https%3A%2F%2Ffleet-api.prd.na.vn.cloud.tesla.com' \
  --header 'Content-Type: application/x-www-form-urlencoded'
```

techguy993, 2024-01-24:
yes tried that...here is the curl command

curl --request POST \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data-urlencode 'grant_type=client_credentials' \
  --data-urlencode "client_id=$CLIENT_ID" \
  --data-urlencode "client_secret=$CLIENT_SECRET" \
  --data-urlencode 'scope=openid vehicle_device_data vehicle_cmds vehicle_charging_cmds' \
  --data-urlencode "audience=$AUDIENCE" \
  'https://auth.tesla.com/oauth2/v3/token'

gilhrpenner, 2024-01-24:
@onlyusedtesla  why has this marked as closed? Did you figure out what was the issue??

onlyusedtesla, 2024-01-24:
It started working for us.

gilhrpenner, 2024-01-24:
> It started working for us.

Did you do anything different? Did you regen your client id and secret or changed anything else?

onlyusedtesla, 2024-01-24:
I work with a partner he's the dev. He told me. I will ask him and get back to you. I do mostly sales. He said tesla updated the backend too and he used the recommendation from @ThomasAlxDmy so it started working.

gilhrpenner, 2024-01-24:
> I work with a partner he's the dev. He told me. I will ask him and get back to you. I do mostly sales. He said tesla updated the backend too and he used the recommendation from @ThomasAlxDmy so it started working.

Do you mind re-opening this issue as some other people are still having issues with it?

onlyusedtesla, 2024-01-24:
done. 
@gilhrpenner

jdolderer, 2024-01-25:
I was using headers before, which doesn't work.

Using it like this works:

```curl --request POST \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data-urlencode 'grant_type=authorization_code' \
  --data-urlencode "client_id=$CLIENT_ID" \
  --data-urlencode "client_secret=$CLIENT_SECRET" \
  --data-urlencode "code=$CODE" \
  --data-urlencode "audience=$AUDIENCE" \
  --data-urlencode "redirect_uri=$CALLBACK" \
  'https://auth.tesla.com/oauth2/v3/token'
  ```
  
  See [documentation](https://developer.tesla.com/docs/fleet-api#authentication)

gilhrpenner, 2024-01-25:
> I was using headers before, which doesn't work.
> 
> Using it like this works:
> 
> ```
>   --header 'Content-Type: application/x-www-form-urlencoded' \
>   --data-urlencode 'grant_type=authorization_code' \
>   --data-urlencode "client_id=$CLIENT_ID" \
>   --data-urlencode "client_secret=$CLIENT_SECRET" \
>   --data-urlencode "code=$CODE" \
>   --data-urlencode "audience=$AUDIENCE" \
>   --data-urlencode "redirect_uri=$CALLBACK" \
>   'https://auth.tesla.com/oauth2/v3/token'
> ```
> 
> See [documentation](https://developer.tesla.com/docs/fleet-api#authentication)

Can confirm this is working now.

techguy993, 2024-01-25:
This is the authorization code request type. The above post in posted is the client credentials request type which does not work. Do you have an example of that request type?

firefixmaarten, 2024-01-26:
+1,
I also get the error when trying to request partner token...
```
# Partner authentication token request
curl --request POST \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data-urlencode 'grant_type=client_credentials' \
  --data-urlencode "client_id=$CLIENT_ID" \
  --data-urlencode "client_secret=$CLIENT_SECRET" \
  --data-urlencode 'scope=openid vehicle_device_data vehicle_cmds vehicle_charging_cmds' \
  --data-urlencode "audience=$AUDIENCE" \
  'https://auth.tesla.com/oauth2/v3/token'
```
gives:
{"error":"invalid_request","error_description":"Invalid resource","error_uri":"https://auth.tesla.com/error/reference/02aacff7-1399-4ce9-bb88-d0f26d480e66-1706258213930"}%                                                                                                                                                                   

To bad the reference does not work...

gilhrpenner, 2024-01-27:
> +1, I also get the error when trying to request partner token...
>
> ```
> # Partner authentication token request
> curl --request POST \
>   --header 'Content-Type: application/x-www-form-urlencoded' \ [...]

With this request I am getting

{"error":"invalid_scope","error_description":"The client is not allowed to access the requested resource.","error_uri":"https://auth.tesla.com/error/reference/d3260b15-4d24-4610-a002-e462e7d180f9-1706315679547"}

ogudet, 2024-02-02:
Hello, does someone has made progress to resolve this issue ?
I'm facing the same error: "error_description": "We don't recognize this client_id and client_secret combination"

ogudet, 2024-02-02:
fond there https://documenter.getpostman.com/view/781424/2s9YRCWB4f#6da852b2-ca25-4d62-b268-fb20384abee0
some node.js code where with the same data as the curl the request is successful.....

gilhrpenner, 2024-02-02:
> fond there https://documenter.getpostman.com/view/781424/2s9YRCWB4f#6da852b2-ca25-4d62-b268-fb20384abee0 some node.js code where with the same data as the curl the request is successful.....

[Shankar's Blog](https://shankarkumarasamy.blog/) is on point and the steps they added for the postman is great. Thanks for this link!!

ymerdrengene, 2024-02-16:
I get the same issue

`{
    "error": "unauthorized_client",
    "error_description": "We don't recognize this client_id and client_secret combination",
    "referenceID": "32705617-9612..."
}`

techguy993, 2024-02-24:
Still doesn't work....

{
  "error": "unauthorized_client",
  "error_description": "We don't recognize this client_id and client_secret combination",
  "referenceID": "f1fa785f-08cb............
}

sethterashima, 2024-03-14:
Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

In order to improve our response times, the scope of the Github issues tracker is being narrowed to the Golang code in the SDK. Backend issues should be reported to the Fleet API support team.

closed 2024-03-14 by sethterashima