teslamotors/vehicle-command #135: Feature Request: Add vehicle year in Fleet API vehicle_data endpoint

sutton, 2024-01-09:
Per the docs, Pre-2021 Model S and X vehicles do not support the new command protocol, and the Fleet API will continue to work for these vehicles. The model is provided for the vehicle in the `car_type` field of the Fleet API [vehicle data endpoint](https://developer.tesla.com/docs/fleet-api#vehicle_data), but the year of the vehicle does not appear to be present in any Fleet API endpoint.

Can the year of the vehicle be added to the vehicle data endpoint so we can easily determine which cars to use the the Command SDK for and which to use the Fleet API for? We are not using the http proxy.

sethterashima, 2024-01-10:
As a work-around, you can determine this dynamically by:

```go
import "github.com/teslamotors/vehicle-command/pkg/protocol"

err := car.StartSession(ctx, nil)
if errors.Is(err, protocol.ErrProtocolNotSupported) {
    // use REST API
}
```

sutton, 2024-01-24:
@sethterashima yes thanks for the suggestion. It would still be helpful to know the year during user onboarding so it is clear if you need to prompt the user to enroll the public key or not. It is also nice metadata to have to display to the user.

closed 2024-06-25 by sethterashima

ruhi-kaluza, 2024-07-01:
Hi @sethterashima ,

I see you've marked this issue as "completed". Does this mean the vehicle year is now available from the Fleet API vehicle_data endpoint?