teslamotors/vehicle-command #137: Unable to authorize 3rd party app

mmrazik, 2024-01-11:
This is really just duplicate of https://github.com/teslamotors/vehicle-command/issues/126 but we were told to create a new issue (shrug).

**Steps to reproduce:**

1. Go to https://tesla.com/_ak/tessie.com
2.  Open it in the mobile -- the Tesla App Opens

**Actual result:**
(another screenshots in english are in the original issue; the content of the error message is the same):

[image]

**Expected result:**
Instead of the above I get a prompt to approve or deny the Oauth.

mmrazik, 2024-01-11:
Or maybe I misunderstood. I'll give it another try before reopening.

closed 2024-01-11 by mmrazik

mmrazik, 2024-01-11:
You are right. Everything works as design. Sorry, I didn't follow the docs properly. 
For others: 
You indeed need to call this first https://auth.tesla.com/oauth2/v3/authorize?&client_id=$CLIENT_ID&locale=en-US&prompt=login&redirect_uri=$REDIRECT_URI&response_type=code&scope=openid%20vehicle_device_data%20offline_access&state=$STATE
 
Prior asking the user to access the tesla.com/_ak/example.com.

yohom, 2024-01-12:
Hello, @mmrazik , I also encountered this problem, but I've been through https://auth.tesla.com/oauth2/v3/authorize apply to the user authorization, I still failed, or the phone key is must first connected when doing this process?

don41382, 2024-03-30:
> 1. https://tesla.com/_ak/tessie.com

Thanks m

> You are right. Everything works as per design. Sorry, I didn't follow the docs properly. For others: You indeed need to call this first https://auth.tesla.com/oauth2/v3/authorize?&client_id=$CLIENT_ID&locale=en-US&prompt=login&redirect_uri=$REDIRECT_URI&response_type=code&scope=openid%20vehicle_device_data%20offline_access&state=$STATE
> 
> Prior asking the user to access the tesla.com/_ak/example.com.

Thanks for pointing this out. I did the same mistake.