teslamotors/vehicle-command #139: Users randomly receiving "User revoked consent" 401 errors

stx, 2024-01-11:
Some users are (randomly?) receiving "User revoked consent" 401 errors whenever their access token is refreshed. This is happening multiple times a day, and it forces them to reauthorize their account over and over again - but the reauthorizations are automatic and do not prompt them for consent, so consent has not actually been revoked.

They claim they are not altering their Tesla account in any way and have no other apps or services running.

Here is an example Tesla OAuth response:

```
HTTP 401
{"error":"login_required","error_description":"User revoked consent","error_uri":"https://auth.tesla.com/error/reference/c9e8ec46-74d8-4eb1-9ff1-caab349db462-1705012163951"}
```

sapkra, 2024-01-17:
I think the problem happens if a user who is signing in is registered in another region than the partner account. My partner account is in the EU but NA users are getting exactly this error.

stx, 2024-01-19:
> I think the problem happens if a user who is signing in is registered in another region than the partner account. My partner account is in the EU but NA users are getting exactly this error.
> 
> After changing the region from one account from NA to EU everything works fine for this account.

It sounds like you found a solution to this, but I don't follow. What do you mean you changed the region?

sapkra, 2024-01-20:
You can move a user to EU servers (the region of the account) in the account settings on the Tesla Website. As a developer you should know how because it's disabled for some reason to "normal" users.

stx, 2024-01-20:
> You can move a user to EU servers (the region of the account) in the account settings on the Tesla Website. As a developer you should know how because it's disabled for some reason to "normal" users.

@sapkra Weird, I can't find this setting anywhere on either user or developer accounts. Can you post a URL and screenshot of this setting?

sapkra, 2024-01-21:
https://accounts.tesla.com/account-settings/personal-information/edit-basic-information

sapkra, 2024-01-24:
Here the full explanation how to change the region:

1. Go to the page [Account Settings](https://accounts.tesla.com/account-settings/personal-information) and click on "Edit Basic Information"
2. Make "Inspect element" on the region select input.
3. Remove disabled and add a new option with the value of your region e.g. `DE` for Germany.
4. You can now select the new region and language in the user interface and save the changes.

jaggerc, 2024-01-26:
It looks like this is related to an issue I just posted as well.  Were you able to figure anything out besides going into the users profile page on Tesla.com?

jaggerc, 2024-01-27:
I am getting the same when refreshing a token.  Data calls work fine until the token expires.

HTTP 401
{"error":"login_required","error_description":"User revoked consent"

It's only for eu users.  We are registered in the US.

We have tried every combination of things to change when registering an eu user but still get stuck on the refresh not working.  Unfortunately we cannot ask every user to go in and change their region in developer mode.

Did anyone figure this issue out?

stx, 2024-01-28:
I'm receiving hundreds of emails about this and really am not sure what to tell people. I have tried pretty much everything imaginable.

jpgough, 2024-01-28:
+1 I’m one of those people emailing @stx as a user. This needs to be fixed, as it’s clearly not implementing the OAuth2 standard correctly from a user perspective

itsMeDavidV, 2024-01-28:
Can confirm we are seeing many cases of this. This should be treated as critical @sethterashima

sp00nsthlm, 2024-01-31:
Hey!

I have this problem with one of my two 3rd party apps, the Smart charging at home connected to my electricity company "Tibber" works perfectly but the Tessie app (for driving record) logs out after exactly 8 hours. Very surprising as I need it for accounting of the car's private versus service miles. Do we know if anyone is working on a solution?

kevin39, 2024-01-31:
Same here. Tessie logout every 8h. very annoying :(
Didn't found any working workaround.

KonradGridio, 2024-01-31:
About 50% of our EU fleet is now having this issue of token's not refreshing and "user revoked consent" 401 (despite evidence that consents are still valid). We are EU-based, and all the fails are cars with customers on EU-accounts. Our Tesla-community is disappointed, after having taken them through the Fleet API onboarding efforts. 

Tesla, please advise. Also happy to provide additional info as needed.

sp00nsthlm, 2024-01-31:
> It happen to me as well. My wifes car was for any reason in region "US". After moving it to "EU" / German account as described by @sapkra it works now. Please Tesla fix the bug!

My account is locked to my country sweden and i can't choose any other? i did however put in my home adress, se if that will change the fault...

fabianhu, 2024-01-31:
Did you define the scope "offline_access" at registration and did the token refresh work?
offline_access enables token refresh without another login by the customer.
Token expiry time is 8h.

I don't know much about the Go technique refreshing tokens, but my python implementation kind of works.
Hope this helps.

jpgough-ms, 2024-02-01:
@stx I'm not sure if this helps, or is just random. As has been usual over the past few days Tessie kicked me out yesterday and asked for an access token. However this time I happened to be in a private browser on Safari on iOS. I had to log back in with username, password, authenticator code. 

Since I've done this I've not been kicked out again (I was expecting that to have happened this morning and yesterday afternoon).

jpgough, 2024-02-01:
@stx I'm not sure if this helps, or is just random. As has been usual over the past few days Tessie kicked me out yesterday and asked for an access token. However this time I happened to be in a private browser on Safari on iOS. I had to log back in with username, password, authenticator code.

Since I've done this I've not been kicked out again (I was expecting that to have happened this morning and yesterday afternoon).

stx, 2024-02-01:
It appears that Tesla has rolled out an auth system update with a new refresh token format, and this problem seems to be fixed now judging from the experience of several users. Can anyone else confirm this issue has been resolved for them as well before I close this?

itsMeDavidV, 2024-02-01:
@stx No I'm still seeing frequent cases of login_required. I'd hold off on closing this.

KonradGridio, 2024-02-01:
We see that Tesla updated refresh_token structure yesterday. Which means all fleet_api token refreshes failed (401 login required - our entire fleet on those), but then after customers log in again, then everything works and the original 401 Error Revoked problem disappears.

So a heavy-handed fix it seems, but did the trick.  Now just have to get all our drivers to re-connect :(.

itsMeDavidV, 2024-02-01:
This is very inconvenient.

sp00nsthlm, 2024-02-01:
Hi, mine seems to have been working since yesterday, had an update to the Tesla app from the App-store in my phone, but so far it's been working and I'm hopeful now. But eagerly waiting for it to crash again…

jaggerc, 2024-02-01:
Just out of curiosity, where are you seeing that Tesla updated the token refresh system?  Are they publishing a log somewhere?

Chrophe, 2024-02-01:
Hi, I don't have to log in to Tesla since yesterday as well. I switched my connectivity in Tessie account yesterday and I had to log in again with my ID and 2FA. No problem anymore.

kevin39, 2024-02-01:
No logout anymore on my side since +30h ago. That never happened.
Maybe the issue is solved ?

sp00nsthlm, 2024-02-01:
How difficult would it be for tesla to create a driving journal themselves to keep track of their private vs business miles, with the option to save as an excel file? then we wouldn't have this problem...

ThomasAlxDmy, 2024-02-01:
Hey all we've now upgraded the system, this bug should be resolved. Thank you for the feedback to chase this down!

stx, 2024-02-01:
@ThomasAlxDmy

Thanks for confirming. I believe those still having issues are running into the new change that refresh tokens can no longer be used twice: https://github.com/teslamotors/vehicle-command/issues/160.

closed 2024-02-01 by stx