teslamotors/vehicle-command #141: auth.tesla.com can erroneously display access revoked message

stx, 2024-01-17:
I realize this repo isn't for the Tesla auth API, but cataloging it here for visibility with Tesla and other developers that I'm sure will run into this.

If you send a user to https://auth.tesla.com/en_us/user/revoke/consent?revoke_client_id=[client_id], it may erroneously show the revoked message automatically without pressing the Revoke Access button. Notably, access is **not** actually revoked.

If the user then visits https://accounts.tesla.com/en_us/account-settings/security?tab=tpty-apps, they can click the app, see that access is not revoked, and then click the Revoke Access button, which does revoke access.

This has happened to several users, and the shared factor is that they're on an EU account. So seems like possibly a weird redirect issue going on there.

stx, 2024-01-17:
I wonder if this is somehow linked to https://github.com/teslamotors/vehicle-command/issues/139?

ThomasAlxDmy, 2024-02-01:
We've now upgraded the system, this should be resolved. Thank you for the feedback to chase this down!

closed 2024-02-07 by derekargueta