teslamotors/vehicle-command #144: Commands via proxy do not accept Fleet API ID

RichieB2B, 2024-01-19:
The [specs](https://developer.tesla.com/docs/fleet-api#charge_start) for `charge_start` specify the `vehicle_tag` as:

> VIN or id field of a vehicle from /api/1/vehicles endpoint.

However using the id field gives this error:

```
{'response': None, 'error': 'expected 17-character VIN in path (do not user Fleet API ID)', 'error_description': ''}
```

The proxy is too strict (and has a typo in the error):
https://github.com/teslamotors/vehicle-command/blob/d7a3e30f4f631b36760c61bc980157ea6de6d2bd/pkg/proxy/proxy.go#L226-L229

RichieB2B, 2024-01-20:
When I patch `proxy.go` to accept the Fleet API ID the command fails with:
```
2024-01-19T23:27:49+01:00 [warn ] [14d97b13b8dea6672c0c68630f7118dc] Session info error: InvalidSignature: session info hmac invalid
2024-01-19T23:27:49+01:00 [debug] [4f00b4e5be7c73153d1c2b81ddf0423a] Retrying transmission after error: Post "https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/929853189081408/signed_command": context canceled
```
Is this a limitation of the proxy? Or does `signed_command` require a VIN on the server side? Either way, this should be documented somewhere.

sethterashima, 2024-01-23:
This is a limitation of the proxy. However, the client needs to know the VIN in order to sign and verify messages. If you want to patch your version, the VIN on [this line](https://github.com/teslamotors/vehicle-command/blob/7c9d6d83ba2fc8cbaaaeeb4d0be0b51734cde87a/pkg/proxy/proxy.go#L307) needs to be "real" VIN, not the Owner API Vehicle ID.

RichieB2B, 2024-01-23:
Thanks! It was easy to fix my app to use the VIN. When #146 is merged it is clearly documented and this issue can be closed.

PR #146 (2024-01-23): tesla-httpy-proxy: Use VIN, not Ower API ID
# Description

Please include a summary of the changes and the related issue.

Fixes #144 

## Type of change

Please select all options that apply to this change:

- [ ] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Bug fix (non-breaking change which fixes an issue)
- [x] Documentation update

# Checklist:

Confirm you have completed the following steps:

- [ ] My code follows the style of this project.
- [ ] I have performed a s [... cut]
closed 2024-01-23 by sethterashima