teslamotors/vehicle-command #148: do you have to be a business to use this?

doubledrat, 2024-01-23:
I'm a hobbyist and have been extensively using the tesla customer api to control my car and my powerwall.
it stopped working last night.
It looks like I can't use the new method unless I'm a tax-registered business?

ThomasAlxDmy, 2024-01-24:
tax-registered business is supposed to be optional. There's a bug right now for people in the UK (we're going to fix that) but  otherwise it should not require tax id. Let me know if this is not true.

sheng168, 2024-01-24:
I was able to register in the US as a indie developer/hobbyist

doubledrat, 2024-01-25:
not in the UK.  needs a number.  if I invent one, then it gives an error
and doesn't register, so I think it's very broken for the UK?

derekargueta, 2024-01-31:
@doubledrat @SebastianGode if you're still experiencing this issue please email me `deargueta at tesla.com` and I can help debug. I've been told it is only required in the UK and if that's not the case I'll need a little assistance tracking down the issue.

andreasfaerber, 2024-02-01:
I was able to submit the request by using DE1234567890 and filling up as much zeros as possible (You can see when you are able to submit the form) for a German account.

SebastianGode, 2024-02-02:
@andreasfaerber 
This is actually lying and should not be done.
Also my request got declined with that method. DId you do it a long time ago? Because at the beginning this was working.

I contacted Derek and will wait for an answer.

andreasfaerber, 2024-02-02:
Not sure if i would call it lying. I have entered personal use as business and the number is obviously invalid. We'll see..

DaveTBlake, 2024-02-02:
> tax-registered business is supposed to be optional. There's a bug right now for people in the UK (we're going to fix that) but otherwise it should not require tax id. Let me know if this is not true.

@ThomasAlxDmy @derekargueta  clearly this has not been fixed yet. I have a valid UK business of many years trading but we are not registered for VAT.  The registration web page requires a taxid of the form `GB123456789`, and I guess that the 9 digits are a VAT number. In the UK businesses have a Unique Taxpayer Reference number (UTR), which is a 10-digit code,  that uniquely identifies the business as registered with the Inland Revenue. This is not the same thing as a VAT number.

Out of frustration I have actually tried a submission with our old VAT number (our current turn over is now too small) just to enter an almost valid taxid and get to the rest of the submission. This still ends at the  "unable to onboard" page, with the browser console showing a "Failed to load resource: the server responded with a status of 400 (Bad Request)" error.

pascalw, 2024-02-03:
Also not working in NL. If I fill in a random taxID the onboarding fails with an HTTP 400 "invalid business" response.

arubinst, 2024-03-15:
Was this solved for people who wrote here over a mont ago?

I am in Switzerland and also required to enter taxID.

SebastianGode, 2024-03-17:
@arubinst No, still no way to register in Germany.
@derekargueta Can you please have a look at the registration form?

Pearse500, 2024-04-09:
I just tried from Ireland and it will not accept my Tax ID, hope this gets resolved soon!!
@derekargueta

cray19003, 2024-05-28:
A TaxID is also required for France. I would like to setup an account to use it in Home Assistant.
@derekargueta Can you have a look at the form please

derekargueta, 2024-05-28:
@cray19003 Sorry I don't work at Tesla anymore, I can't help with this issue.

ThomasAlxDmy, 2024-05-28:
Flows are broken in most of EU currently (forcing a business ID). We're working on a most native way to support developers. 

Coming up in Q3.

liorboyango, 2024-07-07:
Seems like there's a new UI for registration, allowing non-bussiness application :)

closed 2025-01-22 by thefireblade