teslamotors/vehicle-command #153: Fleet API Auth Customer Error

jaggerc, 2024-01-26:
Is anyone getting 'Auth Customer Error' when trying to refresh a token.  We are receiving it on about 10% of users that have switched over to the fleet api.  They have not reset their password or done anything to change the authentication after they authenticated.  Not sure what is going on.

stebo1970, 2024-01-27:
Ich bekomme diesen Fehler:

_Tokenerzeugung fehlgeschlagen

insufficient scopes: missing token scopes: [user_data]_

Jemand eine Idee was ich tun kann?

andig, 2024-01-27:
@stebo1970 your question doesn't have anything to do with this repo and is OT.

jaggerc, 2024-01-27:
Duplicate

closed 2024-01-27 by jaggerc

stebo1970, 2024-01-27:
> @stebo1970 your question doesn't have anything to do with this repo and is OT.

sorry!