teslamotors/vehicle-command #154: Feature Request: Fan Speed

stx, 2024-01-28:
Tesla customers frequently ask to be able to change fan speeds remotely, which is particularly useful while camping. It would be nice to include this in Fleet API + the SDK.

geordi1, 2024-01-28:
Definitely need this function! 

Also: The ability to control the rear AC on dual AC equipped cars - INDEPENDENTLY of the front AC. 

There is NO REASON when the two evaporator coils are valved separately, to require that the front blower be on for the rear blower to be used. The front could remain off and save power when car camping and nobody is sitting in the front seats.

My Model X currently consumes 25-30% of the battery overnight with the rear AC on, where my Model S (same size battery) only consumes around 20% in the same time with climate on at higher settings B/c no rear fan.

eclipsed4utoo, 2024-01-29:
> Definitely need this function!
>
> Also: The ability to control the rear AC on dual AC equipped cars - INDEPENDENTLY of the front AC.
>
> There is NO REASON when the two evaporator coils are valved [...]

This repository is for the Fleet API and/or the Tesla Command SDK. General feature requests for the system is not something that should be done here. You should contact Tesla support or tweet(X'd?) at Elon directly(no, I'm not kidding) for feature requests like this.