teslamotors/vehicle-command #155: Feature Request: In Camping Mode, IGNORE presence of key entirely for safety.

geordi1, 2024-01-28:
This really should have already been done, I'm surprised it wasn't. 

IGNORE THE PRESENCE OF THE KEY - DO NOT AUTOUNLOCK WHEN IN CAMPING MODE.

"walk away locking" is already disabled when in camping mode. WHY then if you manually lock the doors for safety while sleeping (as you should)... WHY would the Model X randomly decide it detects the key and unlock the doors? Sometimes this will happen 3-4 times a night. As I'm asleep I don't know exactly how long it takes, but even with the key under a cellphone so the bluetooth shouldn't detect - it frequently won't find the key when in my POCKET under the phone - it still manages to sense it and unlocks the doors. 

I have used a faraday bag to some success - although the quality of said pouch is suspect and even that wasn't 100% effective. 

Just keep the doors locked when I press the button until I press the button again to unlock! No automation of locks when in camping mode. That's all. Thanks.

rileymd88, 2024-01-28:
Just turn the phone key off when in camp mode?

geordi1, 2024-01-28:
Not on the classic Teslas where you are using an actual key fob that communicates with Bluetooth Low Energy. Phone key doesn’t work on those.Sent from my iPhone

jaggerc, 2024-01-29:
Is your "feature request" issue related to Tesla's vehicle command sdk or the fleet api?

derekargueta, 2024-02-07:
Thanks for the feature request, but this seems very out of scope for this repository. I'll forward this to a product manager but this is not the place to submit general vehicle feature requests and can't guarantee when/if it gets implemented.

closed 2024-02-07 by derekargueta