teslamotors/vehicle-command #156: Can't register key in vehicle

DeLN0, 2024-01-30:
@sethterashima 
Hi 

After i follow all authorization steps, i get the code from the callback url and generate tokens

I open the url from the qr code on my iPhone (https://tesla.com/_ak/https://solarcharger.app), this in turn opens my Tesla app and i get:

[image]

How can i get the key approved in the vehicle?

dmitrybirin, 2024-01-30:
Hello! 👋  @DeLN0 
Try [https://tesla.com/_ak/solarcharger.app](https://tesla.com/_ak/solarcharger.app)
Without https://
if solarcharger.app is the domain you're using for sharing the cert in .well-known section

DeLN0, 2024-01-30:
> Hello! 👋 @DeLN0 Try https://tesla.com/_ak/solarcharger.app Without https:// if solarcharger.app is the domain you're using for sharing the cert in .well-known section

Hi @dmitrybirin , ty for replying

I just tried https://tesla.com/_ak/solarcharger.app on my iPhone but i get the same exact error.

I'm wondering if in previous step, when i opened 
`https://auth.tesla.com/oauth2/v3/authorize?&client_id=REDACTED&locale=en-EU&prompt=login&redirect_uri=https://solarcharger.app/callback&response_type=code&scope=openid%20offline_access%20vehicle_device_data%20vehicle_cmds%20vehicle_charging_cmds&state=0f9d044ff95eb4c0ce08678bb0b4822c`

i should also use solarcharger.app instead of https://solarcharger.app ?

Any other suggestions?

jbanyer, 2024-01-30:
Have you successfully called the [register](https://developer.tesla.com/docs/fleet-api#register) endpoint (using a partner token), and then verified it worked using the [public_key](https://developer.tesla.com/docs/fleet-api#public_key) endpoint?

DeLN0, 2024-01-30:
> Have you successfully called the [register](https://developer.tesla.com/docs/fleet-api#register) endpoint (using a partner token), and then verified it worked using the [public_key](https://developer.tesla.com/docs/fleet-api#public_key) endpoint?

@jbanyer 

This was exactly my problem!
I did call the register endpoint before but i did not remove https:// from the domain and missed the error.
Checking the response from the server it told me to remove https:// from the domain and thats what i did.

Now it works.

Ty @jbanyer !

closed 2024-01-30 by DeLN0