teslamotors/vehicle-command #159: unable to register allowed origin

minhster99, 2024-02-01:
Hi, I'm attempting to register a domain I own https://robee.click but it says 
> Domain must be registered with a certificate authority. 

What does this mean exactly? It's hosted on AWS and has a valid certificate.

[image]

derekargueta, 2024-02-01:
Checking this internally, may be that the Amazon root CA is not in the list of checked authorities.

jbanyer, 2024-02-01:
@derekargueta I have successfully registered an app and configured Allowed Origin URLs including `https://app.chqtest.net`, which uses an Amazon CA cert. This was done some time ago.

I just tried editing the list of Allowed Origin URLs to add `https://api.chqtest.net`, and it worked.

Does this CA restriction only apply when creating a new app?

[image]

minhster99, 2024-02-01:
@jbanyer my https://robee.click site is failing on update. I just attempted your site and it worked. I checked the root CA on yours vs mine and it is slightly different. The CN on mine is `Amazon RSA 2048 M02` , and yours is `Amazon RSA 2048 M01`

@derekargueta any chance we could add the newer AWS root cert as trusted too?

jbanyer, 2024-02-01:
@derekargueta I'll add my 2c: it would be very inconvenient if Amazon CAs are not accepted. AWS provides a very frictionless mechanism for adding and rotating certs when using their services, and I'm guessing a large number of Tesla API customers are going to use them by default. Cheers.

llamafilm, 2024-02-05:
FWIW, I got the exact same error message for a different reason:  My webserver was listening for HTTPS on a nonstandard port.  When I changed it to 443 this error went away.

DaveTBlake, 2024-02-06:
Having difficulty getting with my SSL, perhaps little wonder since the CA is not on that short list of acceptable providers. But is domain validation enough, or is Tesla expecting organisation validation level certification too?

minhster99, 2024-02-12:
Does anyone know which of the acceptable CAs provide free SSL certs? This is a personal project and not sure I am willing to spend a few hundred bucks per year.

llamafilm, 2024-02-12:
Lets Encrypt works for me

minhster99, 2024-02-12:
@llamafilm I just tried LetsEncrypt but I'm still getting the invalid domain error. Is there something on the Tesla side caching the old cert perhaps?

[image]

llamafilm, 2024-02-12:
Not sure.  Maybe they don't allow the .click TLD ?

minhster99, 2024-02-12:
@llamafilm can you give me an example site you got working with LetsEncrypt? I'll compare the certs to see if there's a difference.

.click domain happened to be the cheapest which is why I got it.

llamafilm, 2024-02-12:
Mine is working with https://tesla.llamafilm.com

minhster99, 2024-02-16:
I've got something working. The answer is to have a .com domain and this will work with the AWS signed certificate as well.

@derekargueta I think you should improve the error message because it is currently misleading. Although I don't know why you wouldn't accept any domain as long as it exists. fyi .com domain costs about 4x more than .click domains

paky79, 2024-03-03:
I have the same issue here because at the moment I'm running HA with Duckdns+nginx and duckdns is not recognized as allowed origin Does it mean that I should buy .com DNS and move everything in cloudflare? I saw that Tesla proxy requires anyway nginx. How to solve?

handya, 2024-04-04:
I'm having the same issue on update, using a .app domain and cloudflare for the cert. I added it a few weeks ago and it worked no problem now when I'm trying to update I get "Domain must be registered with a certificate authority". 

I thought it was an issue with .app as others have mentioned but I tried my other .app domains and they worked, not sure why it's suddenly not liking my existing domain.

getget1980, 2024-04-14:
Hello, same problem here, self-hosted domain, with .fr domain name, and using GeoTrust - powered by DigiCert certificate.

Certificate is delivered by Encryption Everywhere DV TLS CA - G2 which seems to be on the VISA list ...

kiberxx, 2024-04-15:
Hello, facing the same error with "com" TLD and LetsEncrypt certificate. Does anyone know the trick how to make it work? On top of the comments above it is not clear what exactly is the problem. 
https://vdubinskii.mooo.com/

getget1980, 2024-04-15:
It worked with Let's encrypt, so for me the problem was GeoTrust - powered by DigiCert certificate.

kiberxx, 2024-04-15:
Yeah, exactly, checked the domain mentioned above with letsEncrypt cert and don't see a significant difference. They use wildcard cert and have a CNAME record, no CAA records. Weird, maybe it is due to I am using freedns. I don't want to play with the paid domains until the issue is clear.

KevM, 2024-06-10:
I just ran into this issue today. My domain (hosted by Vercel) has a valid cert issued by Let's Encrypt.

Sebazzz, 2024-06-10:
Also running into this issue, certificate issued by Let's Encrypt E6, with ISRG Root X1 as root.

jmcollin78, 2024-06-11:
Hello same problem here with a domain in .fr and a Let's Encrypt certificate:

[image]

getget1980, 2024-06-11:
Today it's not working anymore (without any changes)... I'm starting to being bored by all this stiff ....

KevM, 2024-06-11:
My plan is to email support I’ll be needing this down the road.

jmcollin78, 2024-06-12:
I wrote an email to the support at this address: fleetapisupport@tesla.com
I think you can also do the same to hope thinks will change.
I don't see any reason to refuse domain if signed with a valid certificate and root ca.

sethterashima, 2024-06-21:
This scope of this issue tracker is limited to the Golang code in this repository. As @jmcollin78 suggests, please reach out to the Fleet API support team: https://developer.tesla.com/docs/fleet-api#help-and-support.

KevM, 2024-06-22:
Reached out a week ago and still have not heard from them.

jmcollin78, 2024-06-23:
No news also.

ThomasAlxDmy, 2024-06-24:
Hey All,

We officially added support for Let's encrypt. I've checked some of the URL in this issue most certs are expired. Please renew them and try again. 

If you are having issue with Let's encrypt and your certs are valid add your URL to this issue so we can debug and see what's going on. Thank you!

minhster99, 2024-06-25:
Hi @ThomasAlxDmy 

I am the original poster of this issue. Here are my findings 

| URL                 | CA          | Working |
|---------------------|-------------|---------|
| robee.click         | AWS         | No      |
| robee.click         | LetsEncrypt | No      |
| tesla.llamafilm.com | LetsEncrpt  | Yes     |
| robee2.com          | AWS         | Yes     |

Based on the above, both AWS and LetsEncrypt already works however it seems to require a .com domain. If you relax this requirement, it would likely solve a whole bunch of issues for everyone else here.

KevM, 2024-06-25:
What’s weird is on the past I’ve registered a non .com domain without issue. (.rodeo for testing)

ThomasAlxDmy, 2024-06-25:
Yup if set up properly it should work now. Most common issues we've seen:

1. certs are expired.
2. Common Name (CN) in the certificate not matching the URL

firefixmaarten, 2024-06-25:
Hi, how long does it take for the certificate to be rechecked ? As my previous CN was using a wildcard.
It could also be that the dns propagation did not reach your end.

URL: https://tesla.maartenvandenbranden.be/

timcantryn, 2024-06-26:
I can confirm it's not working on a .be domain with unexpired cert with correct CN, is it possible to get a confirmation whether the allowed domains are limited to the .com TLD or if there are certain unallowed TLD's?
Thanks!
[image]
[image]

jmcollin78, 2024-06-29:
Hello, great new I can confirm it is now working with my .fr domain. Thanks for the fix

firefixmaarten, 2024-06-29:
> I can confirm it's not working on a .be domain with unexpired cert with correct CN, is it possible to get a confirmation whether the allowed domains are limited to the .com TLD or if there are certain unallowed TLD's?
> Thanks!
> [image]
> [image]
> 

It works with tesla.maartenvandenbranden.be. I just got It working. Let's encrypt works ;).

Jopand, 2024-06-30:
It is NOT working on valid *.synology.me, *.mywire.org and *.duckdns.org domains 🤔

acanturgut, 2024-06-30:
My firebase hosted site not pass the validation as well. It looks super random my opinion. 

[image]

timcantryn, 2024-07-01:
In my case I still get the error on beforementioned .be domain, it does seem to be quite random.

acanturgut, 2024-07-02:
Contacting with Tesla fix our problem. They added it manually behalf of us.

sgryphon, 2024-07-06:
They seem to have some limitations on which domains they accept (even if the domains are valid, have valid certs, etc.

BTW. The best cert checker I found was Qualys/SSL Labs, i.e. https://www.ssllabs.com/ssltest/analyze.html?d=tesla.gamertheory.net

All the following had Let's Encrypt certs, but only 'www' worked.

* `tesla.gamertheory.net` -- didn't work, note that the host is in the SAN (subject alternative name), not the CN (common name)
* `tesla-auth.gamertheory.net` -- didn't work, for this on the host was the common name
* `www.gamertheory.net` -- did work

sethterashima, 2024-07-08:
@timcantryn I see https://tesla.cantryn.be/ serves a certificate for localhost.localdomain, not tesla.cantryn.be.

@jopand can you provide specific subdomains to test?

@sgryphon `tesla-auth.gamertheory.net` isn't responding for HTTPS connections, I can only connect over HTTP. I'll follow up on SAN vs CN for `tesla.gamertheory.net`, I think that might one cause of the issue.

timcantryn, 2024-07-08:
@sethterashima, what do you mean? This looks fine to me:
[image]

sethterashima, 2024-07-08:
@timcantryn My mistake. Looks like IT doesn't like me visiting tesla.* domains that don't belong to them. :)

sgryphon, 2024-07-09:
> @sgryphon tesla-auth.gamertheory.net isn't responding for HTTPS connections, I can only connect over HTTP. I'll follow up on SAN vs CN for tesla.gamertheory.net, I think that might one cause of the issue.

Well, it does respond to HTTPS for half the internet (see 'Before' result below), but I forgot about legacy internet, and didn't set up the reverse proxy for the new host name, so was only working for IPv6.

Before:
[image]

I have now added tesla-auth to reverse proxy, so legacy connections are possible:

After (now):
[image]

Note that tesla.gamertheory.net is only IPv6, so you do need a modern internet connection to be able to access it. If a work computer is failing, try a mobile phone (or home Internet) ... or just use a checker like Qualys.

[image]

martinzitka, 2024-07-10:
I have a similar problem to @sgryphon above: my domain is still being rejected

[image]

despite getting an A+ SSL rating:

[image]

And yes, the only reason I can think of is the fact I am running the service on Internet and not legacy internet. Honestly I had no intent running this on IPv4.

Could someone from the Tesla team please confirm that their service is currently only supported on legacy IPv4, and if so, if there are any plans to also support IPv6? Thank you!

caglarsu, 2024-07-12:
> Contacting with Tesla fix our problem. They added it manually behalf of us.

I sent an email Tesla more than a week ago and there is no respond :( We are completely stuck and cannot move forward. Very sad story for us.

sethterashima, 2024-07-17:
@martinzitka @sgryphon When users pair the virtual key using the QR code from tesla.com/_ak/example.com, the mobile app will try to fetch your public key from example.com and verify it matches the one Tesla has on record. This way the app doesn't need to trust Tesla's servers for public key distribution. But it does mean that your website needs to be reachable by the mobile app, and so you may want to consider making your site IPv4 accessible.

sgryphon, 2024-07-18:
Okay, so if there is a mobile user whose carrier doesn't support modern connections then we may want to add some legacy support, such as an IPv4 reverse proxy (I use Cloudflare for some of my domains for this). I have a limited user base where I know there is no one in this category, but it may be relevant for some people to consider.

However Tesla may also want to consider that they fully support the current internet version, IPv6.

Most mobile carriers support IPv6, in fact some such my local carrier, Telstra in Australia, have moved to IPv6-only https://www.sidn.nl/en/news-and-blogs/australias-telstra-switches-mobile-users-to-ipv6-only 

There are probably more IPv6-only carriers than IPv4-only carriers, at least for consumer devices. (They do have legacy support using XLAT464 with carrier-grade NAT)

Also, if you have an Apple IOS app, then all apps submitted to the App Store since 01 June *2016* (8 years ago) must support IPv6-only networks -- if your app does not support IPv6-only networks, then it may be in breach. https://developer.apple.com/support/ipv6/

It is probably good to note / give a warning if you detect a site only supports IPv6 about the way the app works if a user only has IPv4, but you should also fully support modern IPv6.

sgryphon, 2024-07-18:
Out of interest I also just checked my local LAN for both my Model 3 and Tesla Wall Charger. I run a dual-stack network.

The devices do not request a DHCPv6 lease, and do not respond to an EUI-64 auto-configured address, although they do respond to IPv4 pings.

For the car:

[image]

For the charger:

[image]

This is not good to not support modern internet and only support legacy protocols. 

There are also some (not many) ISPs that provide IPv6-only (primarily due to the high costs involved in legacy IPv4), so those customers may not be able to use your products if they are IPv4 only.

djballowe, 2024-08-11:
@handya 

I have the exact same issue with a domain that worked a month ago. did this just sort itself out over time?

handya, 2024-08-11:
> @handya
> 
> I have the exact same issue with a domain that worked a month ago. did this just sort itself out over time?

Yes, I tried again a few days later and it started working again 🤷‍♂️

orosys, 2024-08-12:
If the domain is not registered on the Tesla developer site, try clicking the “Next” or “Update” button repeatedly. It should eventually proceed. In my case, please attempt this under the assumption that you can successfully access .well-known/appspecific/com.tesla.3p.public-key.pem.

acanturgut, 2024-09-16:
> Any more help on this? I'm attempting to register the domain `https://www.myteslapower.co.uk/`
> 
> ![image](https://private-user-images.githubusercontent.com/57369759/367942295-dd62e496-af85-44aa-a4db-a118d969fb15.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjY1MjU3OTcsIm5iZiI6MTcyNjUyNTQ5NywicGF0aCI6Ii81NzM2OTc1OS8zNjc5NDIyOTUtZGQ2MmU0OTYtYWY4NS00NGFhLWE0ZGItYTExOGQ5NjlmYjE1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTE2VDIyMjQ1N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYzYjkyNjRhNGMzNDUwNzQ1MTNjYTdmNDY4MjllMzQxM2JhOTc0ZjcyMGMyY2ZkYzNiOWQ5N2Q0YWUxZmY3MWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vpzsdMKqGstvHCa6Ap1or3egwREoqaKzI7lgrHUHNvc)
> 
> I meet the so-called "requirements", e.g CN matching and a valid certificate.
> 
> If anyone could help that would be much appreciated as none of the above solutions worked and I'm kind of stuck. I already contacted Tesla Fleet API support but no respond regarding the actual issue.

Your domain looks fine, better to nagging tesla costumer service. They will add the domain manually eventually. Better to append localhost or other domains which is you desired to use in the future as well on your email.

JjyKsi, 2024-09-19:
Added ionkeep.app couple days ago and worked fine. Came back today to add some testing urls and all of sudden it says that ionkeep.app isn't ok? 

Using Let's Encrypt, CN seems to match, the certificate is valid etc. Any idea what's going on?

Den-88, 2024-09-28:
I am experiencing issues with my websites:

https://teslatrack.tech/
https://fleet.teslatrack.tech/
https://telemetry.teslatrack.tech/
The main site, https://teslatrack.tech/, worked fine after the initial setup, but after a couple of days, it stopped functioning. I have been unable to get any of these sites to work properly.

Could you please check what might be causing the problem?

[image]

getget1980, 2024-09-29:
Hello experiencing same problems with .fr

ThomasAlxDmy, 2024-12-02:
@Den-88 try to remove Tesla from DNS name, there's logic on ourside to ensure customer don't get confused this is a 3rd party.

jman-bunker, 2024-12-03:
I've texted a bit with Tesla. They also suggested that I changed my dns from tesla.<something>.synology.me. So I did. Changed to car.<something>.synology.me and it still did not work. But Tesla just chose to ignore me from that point... Pretty horrible customer service imho

Jopand, 2024-12-03:
I've texted a bit with Tesla. They also suggested that I changed my DNS from tesla.<something>.synology.me. So I did. Changed to car.<something>.synology.me and it still did not work. But Tesla just chose to ignore me from that point... Pretty horrible customer service imho

[5 more replies]