teslamotors/vehicle-command #160: "error": "login_required" when refreshing tokens

andig, 2024-02-01:
Coming from https://github.com/evcc-io/evcc/issues/11987. Using new-style refresh tokens (no longer JWT tokens but prefixed by `NA_` or `EU_`). It seems that it is no longer possible to reuse the refresh token more than once. Trying to do so leads to:

```json
{
  "error": "login_required",
  "error_description": "Login required",
}
```

Did the `auth` endpoint change behavior?

derekargueta, 2024-02-01:
There has been a behavior change, working on getting the docs updated.

itsMeDavidV, 2024-02-01:
> There has been a behavior change that makes refresh tokens no longer reusable, working on getting the docs updated.

To be frank, this is not acceptable. I wish Tesla dev program would announce changes like these in advance. Some of us have thousands of daily active users and unannounced changes like these create significant issues.

andig, 2024-02-02:
https://developer.tesla.com/docs/fleet-api#2024-02-01-auth-access-token-policy-change

zigam, 2024-02-02:
> https://developer.tesla.com/docs/fleet-api#2024-02-01-auth-access-token-policy-change

Can anyone confirm that a new refresh token actually invalidates existing valid-before-expiration access tokens?  I'm not seeing that, just the old refresh token gets invalidated.

andig, 2024-02-02:
You will need to serialise the token refreshes.

fabianhu, 2024-02-03:
Race conditions are not an issue, when doing the token refresh, when it is about to expire, directly before using the token.
I do it like this and have no issues by calling this [`tokens_refresh()`](https://github.com/fabianhu/tesla_api/blob/main/tesla_api_2024.py#L90) in my communication thread before any action.
It checks, if the token is about to expire. If yes, it refreshes.

I only have one thread for communication. If you have multiple, this would need a mutex. (or whatever it is called in your programming language)

In general I think, it is a good security measure to only have one working token at a time. As  a side effect, the token has to be stored at exactly one place in the application to keep track.

andig, 2024-02-03:
> Race conditions are not an issue, when doing the token refresh, when it is about to expire, directly before using the token.

That's exactly what the Go `reuseTokenSource` does:

```go
func (s *reuseTokenSource) Token() (*Token, error) {
	s.mu.Lock()
	defer s.mu.Unlock()
	if s.t.Valid() {
		return s.t, nil
	}
	t, err := s.new.Token()
	if err != nil {
		return nil, err
	}
	t.expiryDelta = s.expiryDelta
	s.t = t
	return t, nil
}

```

jbanyer, 2024-02-03:
@stx we don't have a problem with token refresh concurrency because we perform token refresh as a background task before each token expires (ie every 8 hours), independently of any tasks which use the access tokens. As you say, that does make it difficult to use a distributed architecture for token refresh. It also means we always refresh every token every 8 hours whether it's being used or not.

AFAIK doing a token refresh does not invalidate existing access tokens, which means the access token use does not need to be serialised with the token refresh, ie it can occur concurrently.

stx, 2024-02-03:
I edited my question for clarity - it was more about how the error is unclear in high concurrency scenarios and how we're expected to handle that.

The returned error is just "Login required" so it's not clear if it's due to a reused refresh token or, say, an expired account authorization. You'd want to do different things in each scenario.

You can have a single threaded or distributed mutex system to avoid the error entirely, but the better solution is for a clearer error, IMO.

In my opinion, intuitively, "Login required" makes it seem like access has been revoked rather than the refresh token was already used.

andig, 2024-02-03:
> In my opinion, "Login required" makes it seem like app access has been revoked and that the user has to log in again.

Agreed, the error message is unclear, especially during token refresh. `(refresh_)token_invalidated` would make the underlying issue more clear.

llamafilm, 2024-02-11:
> As part of a security improvement, the acquisition of a new Refresh token will now render the user’s previous Refresh token and its associated Access tokens invalid.

I read this to mean I can only have 1 valid refresh token.  But how am I supposed to exchange the refresh token for an access token? I'm getting the `Login required` error even the first time I try to do this:
```
curl --request POST \
  --header 'Content-Type: application/x-www-form-urlencoded' \
  --data-urlencode 'grant_type=refresh_token' \
  --data-urlencode "client_id=$CLIENT_ID" \
  --data-urlencode "refresh_token=$REFRESH_TOKEN" \
  'https://auth.tesla.com/oauth2/v3/token'
```

andig, 2024-02-11:
> As part of a security improvement, the acquisition of a new Refresh token will now render the user’s previous Refresh token and its associated Access tokens invalid.

@sethterashima this came up in a now-deleted comment. The understanding war "a user can only have a single inflight refresh token at any time". That would be show-stopper as users use different applications with different tokens. As far as I interpreted the change, an access token can only ever have a single active refresh token. That should not forbid multiple access tokens with individual refresh tokens.

Could you kindly confirm this understanding?

rawmean, 2024-02-12:
It seems that if the access_token expires, you cannot use the refresh_token to create a new access_token (even if the refresh_token has not been used previously). Is this the case?

itsMeDavidV, 2024-02-12:
> It seems that if the access_token expires, you cannot use the refresh_token to create a new access_token (even if the refresh_token has not been used previously). Is this the case?

I haven't observed this behavior. I've personally refreshed tokens that have been expires for hours.

With that said, it does seem like once a token is refreshed, its associated access token becomes invalid, even if it's not expired yet, which is an annoying issue when working with multiple devices or even just in a concurrent environment.

Urkman, 2024-02-12:
I also got this problem. When I try to refresh the access token, I also get a new refresh token back. But sometimes this is not a jwt. 
The token then starts with EU_. 
How can I use this token?

andig, 2024-02-12:
@Urkman please don't spam non-related threads. What you got IS the refresh token, just not jwt.

Urkman, 2024-02-12:
Thanks for your answer...

closed 2024-06-25 by sethterashima