teslamotors/vehicle-command #162: Error when registering endpoint in North America, works fine in EU.

vagh141, 2024-02-03:
Based on some Github discussions we have found out that we should call the /register endpoint (https://developer.tesla.com/docs/fleet-api#register) for both eu (https://fleet-api.prd.eu.vn.cloud.tesla.com/) and na (https://fleet-api.prd.na.vn.cloud.tesla.com/). When trying to call eu endpoint it works just fine, however when calling na endpoint we get this error message:

[image]

We are located in Norway and the account is also registered there.

vagh141, 2024-02-03:
The reason we want to register in NA, is because when we try to send a command to a car in NA we get the error: "vehicle rejected request: your public key has not been paired with the vehicle", however in the EU it works just fine. We also have this from the internal tool. 
[image]

zerocooling, 2024-02-18:
> internal tool.

How can you see this homepage ?
I couldnt find it, even I have tried few hours.

closed 2024-03-14 by sethterashima