teslamotors/vehicle-command #164: API endpoint for car image

rawmean, 2024-02-04:
Is there an API endpoint to get the car image?

eclipsed4utoo, 2024-02-05:
There doesn't seem to be, but that would be amazing.

thefireblade, 2025-01-22:
The easiest way to get a vehicle image is to make use of official tesla static assets in the configurator. An example would be:

https://static-assets.tesla.com/configurator/compositor?context=design_studio_2&options=$MT356,$PN01,$W38A,$IPB2&view=STUD_FRONT34&model=m3&size=1920&bkba_opt=2&crop=1300,500,300,300&overlay=0&

At the moment, we don't explicitly provide the option codes that map the colors, wheels configs, etc, but it should be possible given the data you get from vehicle data

closed 2025-01-22 by thefireblade

rawmean, 2025-01-25:
I do not think there is anything in the `vehicle_data` endpoint that provides those option codes that are needed to construct the url

> The easiest way to get a vehicle image is to make use of official tesla static assets in the configurator. An example would be: [...]

I do not think there is anything in the `vehicle_data` endpoint that provides those option codes that are needed to construct the url