teslamotors/vehicle-command #170: Native apps using fleet API are forced to expose Client secret - how to avoid?

g-mocken, 2024-02-06:
If I am not mistaken, the fleet API closely follows https://datatracker.ietf.org/doc/html/rfc6749#section-4.1, which by itself is fine.
However when using this scheme for implementing a native app (e.g. iOS), the app must contain and transmit the client secret, making it easy to extract or intercept it (see https://datatracker.ietf.org/doc/html/rfc8252#section-8.5). The RFC explicitly mentions the risk of client impersonation.

So what is the recommended way for implementing a native app using the current fleet API (if there is any), or will the fleet API be extended to securely support native apps?

sethterashima, 2024-02-13:
We plan on extending the Fleet API to support native apps. [Stay tuned](https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog).

peter-tab, 2024-02-17:
> We plan on extending the Fleet API to support native apps. [Stay tuned](https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog).

Great. Adding PKCE would further improve security… 😉

jonasman, 2024-03-09:
@sethterashima just for planning purposes.
Is this a new SDK that you are preparing or just a universal/deep link to open the official Tesla app to create the new token?

g-mocken, 2024-07-09:
I have just noticed a recent update on the issue: 
https://developer.tesla.com/docs/fleet-api#2024-07-08-open-source-support
https://developer.tesla.com/docs/fleet-api#open-source-token

jbanyer, 2024-07-10:
The new changes for [open-source tokens](https://developer.tesla.com/docs/fleet-api#open-source-token) appear to provide a solution for obtaining API tokens. But what about command signing? That would still require access to a private key corresponding to a public key installed on the vehicle.

sethterashima, 2024-08-07:
> But what about command signing? That would still require access to a private key corresponding to a public key installed on the vehicle.

The app should generate its own key pair the first time it runs, and use https://pkg.go.dev/github.com/teslamotors/vehicle-command@v0.0.2/pkg/vehicle#Vehicle.SendAddKeyRequest.

closed 2024-08-07 by sethterashima

g-mocken, 2025-01-22:
I do not see the "open-source token" anymore in the documentation (only in web archive for mid 2024). Has this feature been revoked again?