teslamotors/vehicle-command #172: Accessing vehicles when outside Cell Network Coverage

fktc, 2024-02-08:
Hi, not sure if this is the right place to ask:

Assuming a vehicle is locked & outside the cell network coverage, e.g. parked in an underground parking. Are there any other options than the originally issued NFC cards to open and access the vehicle?

fktc, 2024-02-11:
@sethterashima I spent some more time thinking about this. I was wondering if the following would be feasible?

1. User registers public key to his vehicle via `https://www.tesla.com/_ak/ `
2. User does not pair or register any BLE device
3. Users vehicle is moved outside cell phone coverage (i.e. won't be reachable via vehicle-command proxy)
4. User downloads a client app to their mobile phone that uses the private key matching the public key of Step 1 above to unlock the vehicle via BLE

Is this feasible? Does unlocking via BLE require any prior pairing of a device with a vehicle or is registering via the /_ak flow sufficient?

sethterashima, 2024-02-13:
Starting with 2023.38, Fleet Manager keys cannot authorize commands over BLE. This allows vehicle owners to immediately cut off 3rd party access even when the car is offline.

However, the vehicle owner can add or invite other drivers to the vehicle; the drivers can then pair their phones over BLE even if the vehicle does not have internet access. See [Fleet API documentation](https://developer.tesla.com/docs/fleet-api#vehicle-endpoints) for how to do this programmatically.

closed 2024-02-13 by sethterashima

fktc, 2024-02-13:
Thanks. If I'm right this has two implications:

1. Invited drivers need a tesla account
2. To pair the phone, the vehicle does need internet access

I'm looking for the most straightforward solution for the following use case:

1. Owner authorizes Guest User to use the vehicle for a specific time period using pin to drive and the ability to lock/unlock vehicle via `vehicle-command` through an app
2. At start of the booking period, "Guest" driver gets no-touch access to the vehicle incl. pin to drive (i.e. no access to a physical NFC card)
3. Guest drives to an area without cell network coverage
4. Guest leaves car
5. Guest is locked out of vehicle and needs to regain access

If I understand right, the only way to prevent this scenario for the guest is to register to tesla & pair their phone via BLE **before** they're potentially locked out of the vehicle?