teslamotors/vehicle-command #174: your public key has not been paired with the vehicle

PrriyaR, 2024-02-09:
Hi 

I am getting following error when i try to invoke the following curl command.

curl --cacert cert.pem --header 'Content-Type: application/json' --header "Authorization: Bearer <token>" -k --data '{}'  "https://localhost:4443/api/1/vehicles/<VIN>/command/flash_lights"

Response:

{"response":null,"error":"vehicle rejected request: your public key has not been paired with the vehicle","error_description":""}

I have registered the public key that generated using **tesla-key-gen** and mapped the third part app with vehicle.

Also started the http proxy using following command

go run tesla-http-proxy/main.go -tls-key key.pem -cert cert.pem -port 4443

I believe the http proxy is not signing with right private key. And vehicle is rejecting. 

Any help is really appreciated.

-Thanks

PrriyaR, 2024-02-09:
https://github.com/teslamotors/vehicle-command/issues/161 Similar issue 

Could some one tell me which private key i should use to start the proxy?

sethterashima, 2024-02-10:
There are two private keys involved:
* The TLS key, which is used by the HTTP proxy to authenticate itself to the HTTP client (e.g., curl). You specify the TLS key with the `-tls-key` argument when starting the proxy.
* The command-authentication key, which is created by `tesla-keygen` and used by the HTTP proxy to authenticate to the vehicle. This key is typically specified in an environment variable.

If you can get within BLE range of the car and run `tesla-command -ble -vin XXX list-keys`, you should see a list of public keys paired with the vehicle.

If you then run `tesla-control -ble -vin xxx lock`, you should see a log line that looks like this:

```
2024-02-09T16:34:59-08:00 [debug] Client public key: <HEX>
```

The hex string should match one of the keys that's on the vehicle. If it does not, then your environment is configured to use a key that hasn't been paired.

If the vehicle's key list includes your public key, then please reply with any errors you see when running the commands below. This will help diagnose the issue.

```
unset TESLA_CACHE_FILE
tesla-control -ble -vin xxx -domain infotainment climate-off
tesla-control -ble -vin xxx -domain vcsec lock 
```

PrriyaR, 2024-02-12:
Yes I am able to pair the public key and it is working as expected. thank you for helping

closed 2024-02-12 by PrriyaR