teslamotors/vehicle-command #176: Not able to connect with fleet API from different region - IN

jaswanthrf, 2024-02-12:
Tried to get the vehicle data after authentication using access_token, getting below error while calling [vet](https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles) API. Any solution?

Steps followed:
1. https://auth.tesla.com/oauth2/v3/authorize 👍
2. https://auth.tesla.com/oauth2/v3/token 👍
3. https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles ❌ 👇 

```
{
  "error": "Account 183156b7d536-4458-9e56-b9a70d90a5b5 must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com, please see https://developer.tesla.com/docs/fleet-api#register"
}
```

patrickdemers6, 2024-02-13:
Have you called the `/api/1/partner_accounts` endpoint with your partner authentication token?

https://developer.tesla.com/docs/fleet-api#register

zerocooling, 2024-02-17:
I have same problem , did you solve it ?

patrickdemers6, 2024-02-17:
Have you called the endpoint the message links to? It completes registration of your developer account.

https://developer.tesla.com/docs/fleet-api#register

zerocooling, 2024-02-17:
> Have you called the endpoint the message links to? It completes registration of your developer account.
> 
> https://developer.tesla.com/docs/fleet-api#register
Yes
1, I put the key, https://<app domain>/.well-known/appspecific/com.tesla.3p.public-key.pem. 
2.  I run this code by using node js.
3. const myHeaders = new Headers();
myHeaders.append("Content-Type", "application/json");
myHeaders.append("Authorization", `Bearer ${process.env.TESLA_API_TOKEN}`);

const body = JSON.stringify({
   "domain": "my domain"
});

const requestOptions = {
   method: 'POST',
   headers: myHeaders,
   body,
   redirect: 'follow'
};

fetch("https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts", requestOptions)
   .then(response => response.json())
   .then(result => console.log(result))
   .catch(error => console.log('error', error));

**Am I wrong ?**

zerocooling, 2024-02-17:
> Have you called the endpoint the message links to? It completes registration of your developer account.
> 
> https://developer.tesla.com/docs/fleet-api#register

Do I need to install Tesla Vehicle Command SDK ?

patrickdemers6, 2024-02-18:
@zerocooling I emailed you. You seem to have not called the endpoint with a partner authentication token.

zerocooling, 2024-02-18:
Can you please take a look , which part I need to fix ?

nst myHeaders = new Headers();
myHeaders.append("Content-Type", "application/json");
myHeaders.append("Authorization", `Bearer ${process.env.AUTHORIZATION_TOKEN}
`);

const body = JSON.stringify({
   "domain": "
https://tesla-watch-app.com/.well-known/appspecific/com.tesla.3p.public-key.pem
."
});

const requestOptions = {
   method: 'POST',
   headers: myHeaders,
   body,
   redirect: 'follow'
};

fetch("https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts",
requestOptions)
   .then(response => response.json())
   .then(result => console.log(result))
   .catch(error => console.log('error', error));

patrickdemers6, 2024-02-18:
- Is the `AUTHORIZATION_TOKEN` coming from a call to `POST https://auth.tesla.com/oauth2/v3/token`?
- Have you confirmed environment variable is being loaded correctly?
- Your domain should just be `tesla-watch-app.com`.
- What response are you getting from your call?

zerocooling, 2024-02-18:
- Is the AUTHORIZATION_TOKEN coming from a call to POST
   https://auth.tesla.com/oauth2/v3/token? -
   {"access_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6InE0dHg3Q1UyYzI2V1BiemwxZjZjanM3QnhzayJ9.eyJpc3MiOiJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92My9udHMiLCJhenAiOiIyMjQ0ZDIxYjZjZWUtNDRhOC04YTE0LTBjZmNmNjBhMWVjNSIsInN1YiI6ImI5ZWY0YjQwLWZjNTktNGM5Ny1iZDQyLWJmNDQxN2Q1NzdlYiIsImF1ZCI6WyJodHRwczovL2ZsZWV0LWFwaS5wcmQubmEudm4uY2xvdWQudGVzbGEuY29tIiwiaHR0cHM6Ly9mbGVldC1hcGkucHJkLmV1LnZuLmNsb3VkLnRlc2xhLmNvbSIsImh0dHBzOi8vYXV0aC50ZXNsYS5jb20vb2F1dGgyL3YzL3VzZXJpbmZvIl0sInNjcCI6WyJ2ZWhpY2xlX2RldmljZV9kYXRhIl0sImFtciI6WyJwd2QiLCJtZmEiLCJvdHAiXSwiZXhwIjoxNzA4MjQ3NzI3LCJpYXQiOjE3MDgyMTg5MjcsIm91X2NvZGUiOiJOQSIsImxvY2FsZSI6ImVuLVVTIiwiYXV0aF90aW1lIjoxNzA4MjE4OTI3fQ.gAjNcjd_OqrFD_Mkh3c7hc0wjzsJI4BUom0wsQVUTn2yLdwb5jXv4jlvjy87aq4k7sq7tkCFBtQYl78rkaulNwBCw7cfZaGyEA24eqBpF2v2zaFKiQw9XKNhlUJhhiKHJLk-fal-WR3Fgl-lBi4C-YO_dRSUZn3tm_-fFKFeaNaMBlC0la3NCdd4NjH-XYVESwOCY5B3sdn6tDeiFyvqjtUaZf11ka71jGOEHKfZLLUS4_aAs_4SQXMpNElghc7oWuVdUa3q6Q--2v65X-5JAP1MM6LW14ZyQD_ikgaRYLFPe79uoD9NnkT7lI2xWBX4zMgNmkWlsZ3GU-h7CcgDcA","id_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6InE0dHg3Q1UyYzI2V1BiemwxZjZjanM3QnhzayJ9.eyJpc3MiOiJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92My9udHMiLCJhdWQiOiIyMjQ0ZDIxYjZjZWUtNDRhOC04YTE0LTBjZmNmNjBhMWVjNSIsInN1YiI6ImI5ZWY0YjQwLWZjNTktNGM5Ny1iZDQyLWJmNDQxN2Q1NzdlYiIsImV4cCI6MTcwODI0NzcyNywiaWF0IjoxNzA4MjE4OTI3LCJhdXRoX3RpbWUiOjE3MDgyMTg5MjcsImFtciI6WyJwd2QiLCJtZmEiLCJvdHAiXSwidXBkYXRlZF9hdCI6MTY1ODM3ODU4MH0.yHFUlM8rzH7o1lvAx3_T2uBUomFsJ7v4uGapnocC_U8lRXiotep6BbniCTjVTyJXNxDmw0bsEVIl7lHXfDPdV_zrCLtu5j2UCFZbwAPeScx1VM_cDh_2SSE9vST-eS-5EvlC5yNIIyzqf9fkGsmkmwro8vSTp6ZLNqSs3cgOWW1RF4KnlUlpVNugTG_xVWo7EHaM8-ViZCJBuuyJcHfe3HBgasft0JjbH2KO8AuOmt-OInLayXXeHlxML_k4kPqvCwuMLWdoihBOnhRlNZqeWmp3t08zEkIM0RzVXSfrgyeVqd4cxqTAd_useVeElt2gRgerGbSLl53Rmw3e4x_VXA","expires_in":28800,"state":"CA","token_type":"Bearer"}

            *  I took the access token*

   - Have you confirmed environment variable is being loaded correctly?
      - Yes
   - Your domain should just be tesla-watch-app.com.
      - I changed "telsa-watch-app.com"
   - What response are you getting from your call?
      -   error: 'Account 2244d21b6cee-44a8-8a14-0cfcf60a1ec5 must be
      registered in the current region
      https://fleet-api.prd.na.vn.cloud.tesla.com, please see
      https://developer.tesla.com/docs/fleet-api#register'

zerocooling, 2024-02-18:
Also, this is post file
 // The data you want to send via POST
        $fields = [
            'grant_type' => 'authorization_code',
            'client_id' => '2244d21b6cee-44a8-8a14-0cfcf60a1ec5',
            'client_secret' => 'ta-secret.Ty5STbtXNq$***',
            'code' => $code,
            'audience' => 'https://fleet-api.prd.na.vn.cloud.tesla.com',
            'redirect_uri' => 'http://localhost:8012/redirect',
            'scope' => 'openid offline_access user_data vehicle_device_data
vehicle_cmds vehicle_charging_cmds'
        ];

        // URL to send the POST request to
        $url = 'https://auth.tesla.com/oauth2/v3/token';

        // Initialize cURL session
        $ch = curl_init($url);

        $headers = [
            'Content-Type: application/x-www-form-urlencoded'
        ];

        // Set cURL options
        curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);
        curl_setopt($ch, CURLOPT_HEADER,$headers);
        curl_setopt($ch, CURLOPT_POST, true);
        curl_setopt($ch, CURLOPT_POSTFIELDS, http_build_query($fields));

        // Execute the POST request
        $response = curl_exec($ch);

zerocooling, 2024-02-18:
1.Is the AUTHORIZATION_TOKEN coming from a call to POST https://auth.tesla.com/oauth2/v3/token?
   token is fine
2.Have you confirmed environment variable is being loaded correctly?
   I checked it
3.Your domain should just be tesla-watch-app.com.
  ok
4.What response are you getting from your call?
I still got the same error, even I changed us, eu, and cn.
{
  error: 'Account 2244d21b6cee-44a8-8a14-0cfcf60a1ec5 must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com, please see https://developer.tesla.com/docs/fleet-api#register'
}

patrickdemers6, 2024-02-18:
Simple script I just verified to be working. The curl snippets in documentation also work just fine.

```js
const TOKEN_URL = 'https://auth.tesla.com/oauth2/v3/token'
const REGISTER_URL = 'https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts';

const { CLIENT_ID, CLIENT_SECRET, AUDIENCE, DOMAIN } = process.env;

const run = async () => {
    const urlencoded = new URLSearchParams();
    urlencoded.append("scope", "openid vehicle_device_data vehicle_cmds offline_access");
    urlencoded.append("audience", AUDIENCE);
    urlencoded.append("grant_type", "client_credentials");
    urlencoded.append("client_id", CLIENT_ID);
    urlencoded.append("client_secret", CLIENT_SECRET);

    const {access_token} = await fetch(TOKEN_URL, {
        body: urlencoded,
        method: 'POST',
        headers: {
            'Content-Type': 'application/x-www-form-urlencoded'
        }
    }).then(res => res.json());
    
    const res = await fetch(REGISTER_URL, {
        body: JSON.stringify({domain: DOMAIN}),
        method: 'POST',
        headers: {
            'Content-Type': 'application/json',
            'Authorization': `Bearer ${access_token}`
        },
    }).then(res => res.json());
    console.log(res);
};

run();

```

patrickdemers6, 2024-02-20:
Confirmed via email the issue is resolved.

gilbertorobles24, 2024-03-01:
I am still having this issue. I am using AWS API Gateway from us-west-2 (Oregon), and I receive the same error as OP: 

`{
    "error": "Account cf974e2618fb-4a7d-8ddb-52b9e166b6c1 must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com, please see https://developer.tesla.com/docs/fleet-api#register"
}`

I am using a valid authorization token, with proper headers, and my public key is hosted on my domain at the `/.well-known/appspecific/com.tesla.3p.public-key.pem` endpoint. 

I receive the error upon attempting to register, submitting a POST request to https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/partner_accounts

For some reason GET https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/dx/charging/history is the only request that returns the requested information. Any other vehicle-related request returns the region error. I can create and refresh tokens just fine.

Help would be greatly appreciated.

gilbertorobles24, 2024-03-01:
> @gilbertorobles24 please make sure you call the register endpoint using a partner account token.

It works! Yeah I went back to check the docs for the "partner authentication token". Apparently I was using a "Third Party Token". The difference was that third party token is generated with grant type `authorization_code` and partner authentication token for registration is generated using `client_credentials` grant type. Additionally, the register endpoint requires setting header `Accept: application/json`. Thanks a lot!

algology, 2024-03-29:
So if using a local host you wont be able to register an account as the allowed origin will have to be able to server a public key. I'm currently hosting my public key on github but it looks like that wont work as github is a different domain to localhost.

closed 2024-06-25 by sethterashima

CodeMaddin, 2024-08-29:
> Simple script I just verified to be working. The curl snippets in documentation also work just fine.
>
> ```js
> const TOKEN_URL = 'https://auth.tesla.com/oauth2/v3/token'
> const REGISTER_URL = [...]

This was very helpful, thank you!