teslamotors/vehicle-command #178: `vehicle_data` endpoint does not return the cached data if the car is asleep

rawmean, 2024-02-13:
New problem today: `vehicle_data` endpoint does not return the cached data if the car is asleep

```
Optional(https://fleet-api.prd.na.vn.cloud.tesla.com/api/1/vehicles/<vid>/vehicle_data?endpoints=climate_state3Bcharge_state3Blocation_data3Bgui_settings3Bvehicle_config3Bvehicle_state3Bdrive_state)
STATUS CODE: 408

RESPONSE BODY ERROR: {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
```

rawmean, 2024-02-13:
Being able to retrieve the data without waking the car is an important feature. This happened about a month ago also but was fixed after one day. I hope this is a temporary glitch.

rawmean, 2024-02-15:
@derekargueta , @sethterashima can you please indicate if this is a bug or cached data won't be returned anymore?
I wish there was more heads up wrt to changes like this so that developers can handle it more gracefully. No response from the fleet support email either.

ThomasAlxDmy, 2024-04-11:
This is a bug.

rawmean, 2024-04-13:
> This is a documentation issue. We should fix.

The server used to return the cached data just like the documentation says. So, I don't think this is a documentation issue. Are you saying that the behavior has changed and cached data is no longer returned?

Urkman, 2024-06-28:
This is problem still exists...

thefireblade, 2025-01-23:
The issue here is intentional, we're moving forward with making sure the data we provide via the `vehicle_data` endpoint is always up to date and we're no longer serving the cached vehicle data through Fleet API.

thefireblade, 2025-01-23:
Please see https://developer.tesla.com/docs/fleet-api/endpoints/vehicle-endpoints#vehicle-data

Please refer to wording
```
Makes a live call to the vehicle to fetch realtime information.
```

closed 2025-01-23 by thefireblade