teslamotors/vehicle-command #183: BLE protocol: Action HvacTemperatureAdjustmentAction ignores set properties values

hleao, 2024-02-15:
When using the ble protocol to send the action `HvacTemperatureAdjustmentAction`, only values for properties `driver_temp_celsius` and `passenger_temp_celsius` are being considered and all others are ignored.

For example, sending a message with values only for `delta_celsius` = 1 and `hvac_temperature_zone` = 2 (TEMP_ZONE_FRONT_LEFT) does not work. The car ignores the values for those 2 properties and, since `driver_temp_celsius` and `passenger_temp_celsius` are not set, it sets both temperatures to 'LO'.

That makes it is not possible to increase or decrease the temperature, with a given delta, using ble.

hleao, 2024-02-16:
Here is an example, a slightly modified version of `Vehicle.ChangeClimateTemp(...)` from what we have in [climate.go](https://github.com/teslamotors/vehicle-command/blob/9bd8e344e75d845a5db41f9f5ccdaeecb1c469a2/pkg/vehicle/climate.go#L88). Executing this action results in temperature for both driver and passenger being set to 'LO'.

```go
func (v *Vehicle) ChangeClimateTemp(ctx context.Context, driverCelsius float32, passengerCelsius float32) error {
  var zones []*carserver.HvacTemperatureAdjustmentAction_HvacTemperatureZone

  var zone *carserver.HvacTemperatureAdjustmentAction_HvacTemperatureZone =  new(carserver.HvacTemperatureAdjustmentAction_HvacTemperatureZone)
  zone.Type = &carserver.HvacTemperatureAdjustmentAction_HvacTemperatureZone_TEMP_ZONE_FRONT_LEFT{}
  
  zones = append(zones, zone)
  
  return v.executeCarServerAction(ctx,
    &carserver.Action_VehicleAction{
      VehicleAction: &carserver.VehicleAction{
        VehicleActionMsg: &carserver.VehicleAction_HvacTemperatureAdjustmentAction{
          HvacTemperatureAdjustmentAction: &carserver.HvacTemperatureAdjustmentAction{
            DeltaCelsius: 1, // <- Added but no effect 
            HvacTemperatureZone: zones, // <- Added but no effect
            Level: &carserver.HvacTemperatureAdjustmentAction_Temperature{
              Type: &carserver.HvacTemperatureAdjustmentAction_Temperature_TEMP_MAX{},
            },
          },
        },
      },
    })
}
```

sethterashima, 2024-02-20:
This doesn't appear to be a BLE issue. The API defined in climate.go doesn't support those fields because they are currently not used by the car.

@sheng168 In general their should be parity between BLE and HTTP. The commands end up in the same place. There are a small number of exceptions related to security (e.g., fleet manager keys cannot authorize commands over BLE) or technical constraints (e.g., vehicle wake behavior).

closed 2024-02-20 by sethterashima