teslamotors/vehicle-command #185: Backend randomly returns 403

bennibau, 2024-02-16:
Hi,
we have the new Tesla integration running in production and it works so far. Wen have about 600 cars connected as of right now.
Tonight, and also in the past the Tesla backend returned 403 on normal status requests randomly. This causes in our backend the issue, that we assume the user has not enough scopes provided to us. The documentation states that if you get 403, there are scopes missing. The result ist, that we show to our users that scopes are missing and we cannot control the car anymore. However if we just reset the state and try again it works, as before.
How can I determine if the user actually did not provide enough scope or revoked them or if the backend randomly returns 403? I cannot find anything in the documentation or the code of this project.

rileymd88, 2024-02-16:
This does not completely solve the issue, however the token which gets generated is a JWT token which can be decoded to give you the scopes the user selected.

bennibau, 2024-02-16:
But that does not really solve the problem because the backend returns 403. If I decode the token, the user has in our backend I can see the scopes that were provided in the past. But the user always has the possibility to revoke the scopes so I cannot assume that the scopes are still the same.

jbanyer, 2024-02-16:
FWIW, we (chargehq.net) have thousands of vehicles connected and haven't seen this issue.

jbanyer, 2024-02-16:
Also, AFAIK the user can not revoke the scopes of an access token. If they remove third party app permission entirely in tesla.com, the access tokens remain valid and working. I just tested it. I assume they remain valid until they expire.

If you then attempt a token refresh after permission has been revoked you receive a 401.

bennibau, 2024-02-16:
> Also, AFAIK the user can not revoke the scopes of an access token. If they remove third party app permission entirely in tesla.com, the access tokens remain valid and working. I just tested it. I [...]

Yes that is correct but actually I think it is a bug. They should revoke the access tokens when a user changes his scopes. 

But you mean that if the user already had the scopes he cannot run into a 403 because of missing scopes again because he would always get a 401 instead?

jbanyer, 2024-02-16:
> Yes that is correct but actually I think it is a bug. They should revoke the access tokens when a user changes his scopes.
> 
> But you mean that if the user already had the scopes he cannot run into a 403 because of missing scopes again because he would always get a 401 instead?

A user can not change their scopes as such. All they can do is revoke access entirely, then grant permission again, potentially selecting different scopes.

I've just tried this:
- perform permission grant, selecting all scopes
- obtain refresh token R1 and access token A1
- access token A1 is working normally
- revoke third party app access in tesla.com
- access token A1 continues working normally
- attempt token refresh with R1 - returns 401
- perform permission grant again, selecting the incorrect scopes
- obtain refresh token R2 and access token A2
- access token A2 returns http 403, as expected, as scopes are missing
- access token A1 continues working normally

So, as you first said, either your users are changing their scopes (by revoking access and re-adding with the wrong scopes) or there is a bug in the API. We haven't seen such a bug. Most of our users are in Australia (NA region).

bennibau, 2024-02-16:
> > Yes that is correct but actually I think it is a bug. They should revoke the access tokens when a user changes his scopes.
> > But you mean that if the user already had the scopes he cannot run into a 403 because of missing scopes again because he would always get a 401 instead?
> 
> A user can not change their scopes as such. All they can do is revoke access entirely, then grant permission again, potentially selecting different scopes.
> 
> I've just tried this:
> 
> * perform permission grant, selecting all scopes
> * obtain refresh token R1 and access token A1
> * access token A1 is working normally
> * revoke third party app access in tesla.com
> * access token A1 continues working normally
> * attempt token refresh with R1 - returns 401, as expected since user has revoked access
> * perform permission grant again, selecting the incorrect scopes
> * obtain refresh token R2 and access token A2
> * access token A2 returns http 403, as expected, as scopes are missing
> * access token A1 continues working normally
> 
> So, as you first said, either your users are changing their scopes (by revoking access and re-adding with the wrong scopes) or there is a bug in the API. We haven't seen such a bug. Most of our users are in Australia (NA region).

I will have to check the scopes as mentioned by @rileymd88 just in this case.

I think we should file a bug for the mentioned behavior. I think it is not wanted that access tokens are still valid after a user revoked his scopes. I think the accessToken should be revoked the same way as the refresh token is revoked.

sethterashima, 2024-03-14:
Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

In order to improve our response times, the scope of the Github issues tracker is being narrowed to the Golang code in the SDK. Backend issues should be reported to the Fleet API support team.

closed 2024-03-14 by sethterashima