teslamotors/vehicle-command #186: Proxy returns HTTP 500 when sending command to sleeping vehicle

andig, 2024-02-16:
Note this log for retrieving data vs. sending commands (coming from https://github.com/evcc-io/evcc/issues/12294#issuecomment-1948394229):

```
[tesla-proxy] TRACE 2024/02/16 11:28:24 GET https://tesla.evcc.io/api/1/vehicles/xxx/vehicle_data
[tesla-proxy] TRACE 2024/02/16 11:28:24 {"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[lp-1  ] ERROR 2024/02/16 11:28:24 vehicle odometer: asleep

[tesla-proxy] TRACE 2024/02/16 11:29:54 POST https://tesla.evcc.io/api/1/vehicles/xxx/command/set_charging_amps
[tesla-proxy] TRACE 2024/02/16 11:29:56 {"charging_amps": 32}
--
{"response":null,"error":"vehicle unavailable: vehicle is offline or asleep","error_description":""}
[lp-1  ] ERROR 2024/02/16 11:29:56 max charge current 32A: 500 Internal Server Error
```

Sending commands to proxy results in HTTP 500 (still need to verify behaviour for using fleet api directly). Imho it should return HTTP 200 and indicate the error in the body same as with executing `GET` requests.

rawmean, 2024-02-16:
I think the problem is that the vehicle is asleep. Until a few days ago, Tesla server returned cached data when the vehicle was asleep. Somehow they stopped doing that. I'm not sure if this is a bug or if they change the behavior.

andig, 2024-02-17:
This issue is about vehicle asleep for GET requests returning HTTP 200 and for POST requests HTTP 500.

rawmean, 2024-02-17:
> This is NOT what this issue is about. This issue is about vehicle asleep for GET requests returning HTTP 200 and for POST requests HTTP 500. The bahaviour is inconsistent.

I see. But, I get HTTP 408 when I send a `vehicle_data` request. Don't you? I don't send the requests using the proxy. Maybe that's why?

thefireblade, 2025-01-22:
I don't think this request should be returning 200 otherwise you will be charged for requests that are failing. I am going to close this as this is an old issue. Open a new issue if this is still happening

closed 2025-01-22 by thefireblade

traxmaster666, 2025-01-26:
for All who uses EVCC 0.133 (Tesla/TWC3 + Home Assistant): For me, a temporary workaround is using an automation in Home Assistant, which checks EVCC mode switch (-> evcc Integration) and switch on the Sentry mode, each time the charge mode ist changing. This results in enabling the sentry mode, which didn´t harm but enables EVCC to work again without the disturbing "500-errors".

andig, 2025-01-26:
@traxmaster666 this is NOT the evcc repo