teslamotors/vehicle-command #187: Fleet Accounts Error 401

jaggerc, 2024-02-16:
We have a few users with municipalities and their vehicles are under a Tesla Fleet Account with multiple vehicles.  We were able to obtain data from these fleet accounts with the owner api but receive the following error with the fleet api:

{"response":null,"error":"unknown user","error_description":""} ) 401

According to the Fleet API Docs this is due to mobile access being disabled or an expired token but they are able to access their vehicles with the mobile Tesla app and we're able to refresh their token so neither seem to be the issue.

Is there anything we need to change or they need to change to get this working or is the Fleet API not setup for these accounts yet?

rawmean, 2024-02-16:
See this:
`As part of a security improvement, the acquisition of a new Refresh token will now render the user’s previous Refresh token and its associated Access tokens invalid`
From: https://developer.tesla.com/docs/fleet-api#announcements-amp-api-changelog

This is per partner account. so, Tesla app working is not related.

DeLN0, 2024-02-20:
> We have a few users with municipalities and their vehicles are under a Tesla Fleet Account with multiple vehicles. We were able to obtain data from these fleet accounts with the owner api but [...]

This was a bug on Tesla's side that I also experienced. 
Today it works fine.

sethterashima, 2024-03-14:
Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

In order to improve our response times, the scope of the Github issues tracker is being narrowed to the Golang code in the SDK. Backend issues should be reported to the Fleet API support team.

closed 2024-03-14 by sethterashima