teslamotors/vehicle-command #188: Missing proxy commands

Urkman, 2024-02-16:
Are there any plans to add the missing commands to the proxy?

fabianhu, 2024-02-18:
I already added a missing command for [tesla-command](https://github.com/teslamotors/vehicle-command/pull/34).  
Should not be too difficult to do the same for the proxy.
Which commands are you missing?

Urkman, 2024-02-19:
mostly missing "vent windows"

Lenart12, 2025-01-19:
Current state of commands:
### Missing commands
- actuate_trunk
- clear_pin_to_drive_admin
- media_next_fav
- media_next_track
- media_prev_fav
- media_prev_track
- media_volume_down
- navigation_gps_request
- navigation_sc_request
- navigation_waypoints_request
- remote_auto_steering_wheel_heat_climate_request
- remote_steering_wheel_heat_level_request
- speed_limit_clear_pin_admin
- sun_roof_control
- upcoming_calendar_entries
### Undocumented commands
+ close_tonneau
+ open_tonneau
+ set_managed_charge_current_request
+ set_managed_charger_location
+ set_managed_scheduled_charging_time
+ stop_tonneau
+ wake_up *<-- vehicle endpoint, not vehicle command*

vbarrier, 2025-02-25:
Missing clear_pin_to_drive_admin is blocker for me. One of my customers still use the car but doesn't pay anymore, and I can't block him..