teslamotors/vehicle-command #189: App not registered error.

Urkman, 2024-02-18:
Some of my user getting a "App not registered" error, when trying to register the with their teslas.
See screenshot.
[image]

But my app is registered and active.
See other screenshot
[image]

What is the problem here?

Urkman, 2024-02-20:
@sethterashima 
Can someone help me please on this?
This was working fine, but stopped a couple of days ago :(

Urkman, 2024-02-20:
What I did to register my private and public key:
1.) use tesla-keygen: `tesla-keygen -f -key-file private.pem create > public.pem`
Then I have two files:
public
```
-----BEGIN PUBLIC KEY-----
xxx
-----END PUBLIC KEY-----
```
private:
```
-----BEGIN EC PRIVATE KEY-----
yyy
-----END EC PRIVATE KEY-----
```

2.) Copy public file to my server at: `https://domain.app/.well-known/appspecific/com.tesla.3p.public-key.pem`

3.) Register the key using the new API

4.) Ready... And this was working, but stopped working some days ago.

But, when I try to use the new check script at https://github.com/teslamotors/fleet-telemetry/blob/main/tools/check_csr.sh
I get an error:
```
C0BAE5EC01000000:error:0480006C:PEM routines:get_name:no start line:crypto/pem/pem_lib.c:763:Expecting: CERTIFICATE REQUEST
Unable to load X509 request
-e Error: the CSR public key does not match the com.tesla.3p.public-key.pem
```

Someone with an idea, what is wrong here?

Urkman, 2024-02-20:
Now I created a csr file using:
`openssl req -out partner_domain.com.csr -key private_key.pem -subj /CN=[partner_domain.com/](http://partner_domain.com/) -new
`
With this file the check script is fine

sethterashima, 2024-02-20:
Could verify these two hex strings match?

One from your public key:

```
openssl ec -in public_key.pem -pubin -outform DER | tail -c 65 | xxd -p -c130
```

And one from the public key currently registered with Tesla: https://developer.tesla.com/docs/fleet-api#public_key

Urkman, 2024-02-20:
Got an answer from the Tesla fleet support email :)

This was the solution:

> Found the problem. During the key pairing process, the mobile app requests the public key from your server. This request from the mobile app includes the Range header set to “bytes=0-200”. When this header is set, your web server returns 400 Bad Request.

As my backend is written in Vapor/Swift, the Range header was not supported correctly.
I needed to handle this using Nginx :)

closed 2024-02-21 by sethterashima