teslamotors/vehicle-command #190: Howto update scopes of partner account?

andig, 2024-02-19:
After our application ([evcc.io](https://evcc.io)) has evolved it turns out we'd need the `energy_cmds` scope in addition to what we've setup sofar. Right now we cannot create customer tokens including `energy_cmds` since it's not part of our partner account's scopes.

How can we add new scopes to an existing partner account?

DeLN0, 2024-02-20:
> After our application ([evcc.io](https://evcc.io)) has evolved it turns out we'd need the `energy_cmds` scope in addition to what we've setup sofar. Right now we cannot create customer tokens [...]

I don't think you can do that right now.
A workaround would be to apply for a new app with the same name (with the  correct scopes) in your developer account, then use the client id and client secret from the new app.

rkobylinski, 2024-02-23:
That would be useful. Also removing scopes if it turns out you don't need them (yet?) in production.

derekargueta, 2024-02-26:
This is not yet supported but the feedback is being passed along. Closing as a dupe of https://github.com/teslamotors/vehicle-command/issues/74

closed 2024-02-26 by derekargueta