teslamotors/vehicle-command #191: Why is owner allow to lock/unlock but not drive over BLE transport?

sheng168, 2024-02-19:
Is this a bug or by design?
jin@jins-macbook vehicle-command % tesla-control -ble drive 
Failed to execute command: vcsec could not execute command: genericError:GENERICERROR_NOT_ALLOWED_OVER_TRANSPORT
jin@jins-macbook vehicle-command % tesla-control -ble lock        
jin@jins-macbook vehicle-command % tesla-control -ble unlock
jin@jins-macbook vehicle-command %

sethterashima, 2024-02-20:
This is by design. The limitation is currently required to prevent a PIN-to-Drive bypass. Hopefully we'll be able to remove that limitation and/or implement passive authentication in the near future.

closed 2024-02-20 by sethterashima

sheng168, 2024-02-21:
The same command sent over REST is allow to bypass PIN-to-Drive, right? Also, is there any documentation on the passive phone-key protocol?