teslamotors/vehicle-command #192: Access VIN over BLE

sheng168, 2024-02-21:
I working on a stand-alone Apple Watch Bluetooth app. I can add my public key to the vehicle, but the VIN is needed to execute commands and I've found no place to retrieve it. It's not very user friendly to ask user to enter 17 characters of letters and numbers. Is there a better way?

sheng168, 2024-02-21:
I'm putting VIN in vehicle name during the inital setup as a temporary workaround

rileymd88, 2024-02-21:
The following vehicle endpoint will give you a list of vehicles and include
the VIN: api/1/vehicles

closed 2024-02-21 by sethterashima

sheng168, 2024-02-21:
I'm trying to make an app that works when there's no cellular nor wifi; using just BLE

Whirlwind, 2024-03-03:
I have same problem. There's no cellular nor wifi, too. :)

sheng168, 2024-10-08:
I ask the user to add their VIN in the car's name