teslamotors/vehicle-command #193: Getting the private key out of macOS login keychain non-trivial

rkobylinski, 2024-02-22:
After a full restore from a Time Machine backup the private key generated by tesla-keygen seemed gone. While I was able to recover a copy of the login keychain from an older backup manually, I figured it would make sense to get it out of the macOS keychain. In pem format ideally, so you could use it with tesla-keygen migrate if needed. Well this proved possible but not trivial, the following seems to work for me:

``% echo 30310201010420 `security find-generic-password -a vehicleCommandKey.$TESLA_KEY_NAME -w` a00a06082a8648ce3d030107 |  xxd -r -p | openssl ec -inform d > private_key.pem``

Puh, took me a while to figure that out. Or am I missing an obvious easier way?

andig, 2024-03-13:
@rkobylinski same issue here. Could you kindly explain the magic constants in your approach?

rkobylinski, 2024-03-13:
> @rkobylinski same issue here. Could you kindly explain the magic constants in your approach?

My understanding is that these are requirements of the DER / ASN.1 binary format. I found the following post and the included links useful: https://stackoverflow.com/questions/48101258/how-to-convert-an-ecdsa-key-to-pem-format

Note that Tesla uses secp256r1 curve (prime256v1), so the initial example in the answer won't work. A later addition to the answer explains how to adapt pre_string and the mid_string though. Because we are not including the public key here, you have to change the length in the second byte of the pre_string to the length of the following sequence: decimal 49 or hex 31.

PR #259 (2024-06-17): Add export option to tesla-keygen
# Description

Having exportable private keys isn't ideal, but is sometimes needed. This adds an "export" subcommand to tesla-keygen.

Fixes #193 #258 

## Type of change

Please select all options that apply to this change:

- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [ ] Bug fix (non-breaking change which fixes an issue)
- [ ] Documentation update

# Checklist:

Confirm you have completed the following steps:

- [x] My code follows th [... cut]
closed 2024-06-17 by sethterashima