teslamotors/vehicle-command #194: error : You must provide a private key with -key-name or -key-file to execute this command

kurb77, 2024-02-23:
hello,
since few hours, i can't execute any command except wake.
i have this error:
`You must provide a private key with -key-name or -key-file to execute this command`

my command line is:
`./lib/tesla_api/tesla-control/tesla-control -key-file ./lib/tesla_api/TeslaKeys/privatekey.pem -token-file .temp_token -debug -vin XP7XXXXXXNBXXX123 charging-start'
`
before, all worked good, i could change amps, stop and start the charge

with debug arrg, i have this:
no error, Server returned 200: OK:
except the last line:
`2024-02-23T13:12:00+01:00 [warn ] No session available for DOMAIN_INFOTAINMENT
`
thank you for your help

jerome

sethterashima, 2024-02-28:
It looks like the problem is that the client ignores handshake responses from the server if there's too much latency. This is by design. But it still behaves as though the server completed its handshake and continues, which is a bug.

You can increase [these two timeout values](https://github.com/teslamotors/vehicle-command/blob/main/internal/dispatcher/dispatcher.go#L21-L22) as a work-around until the bug is fixed. They should probably be higher, anyway.

Thanks for your report!

kurb77, 2024-02-29:
Hello, i have increase these Two timeout values (5 -> 15), it's ok, i can again send orders to my car

thank you
jerome

PR #207 (2024-03-08): Add timeout options
# Description

Fixes an issue where a handshake would incorrectly be marked as completed if the response came in after the timeout expired.

Adds API and CLI options for adjusting timeouts.

Fixes #194 

## Type of change

Please select all options that apply to this change:

- [x] New feature (non-breaking change which adds functionality)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected)
- [x] Bug fix (non-breaking change which fixes an issue)
- [x] Documentation update

# Checklist:

Confirm you have completed the follo [... cut]
closed 2024-03-08 by sethterashima