teslamotors/vehicle-command #195: Fleet API down?

stx, 2024-02-24:
It seems like all requests to all Fleet API endpoints are failing since 4:30PM PST:

HTTP Error Code 412
{"error":"Account [client-id] must be registered in the current region https://fleet-api.prd.na.vn.cloud.tesla.com/, please see https://developer.tesla.com/docs/fleet-api#register"}

jbanyer, 2024-02-24:
FWIW we (chargehq.net) are not seeing this error. Most of our vehicles are in the NA region.

stx, 2024-02-24:
Okay, so after Tesla completed our fleet-telemetry CSR setup today, the whole account became unregistered from the Fleet API regions.

We fixed this by manually reregistering with each region following the instructions at https://developer.tesla.com/docs/fleet-api#register.

FYI for everyone else that runs into this.

closed 2024-02-24 by stx

jbanyer, 2024-03-04:
I had this happen with our test domain `tesla.chqtest.net`. It occurred many days after Tesla approved the CSR, so that doesn't appear to be the cause.

When it happened, the following changes had been made:
- our public key had been taken down temporarily for about 24 hours
- we enabled fleet telemetry on a vehicle for the first time

So I'm not sure what triggered the issue. Perhaps the public key being taken down?

The fix was to re-register the public key again.