teslamotors/vehicle-command #197: Documentation on errors

robin-karlsson, 2024-02-28:
This night my car failed to start charging multiple times, and looking through the logs I can see this being returned as response from the `signed_command call`:

`{ "operationStatus": "OPERATIONSTATUS_ERROR", "signedMessageFault": "MESSAGEFAULT_ERROR_INVALID_TOKEN_OR_COUNTER" }`

Tried to search the repo for any documentation of what the error is about, and how we can make sure to recover from those errors.

Charging worked when I started it this morning, and it worked for multiple other users on the same service. Tried to increase the size of session cache, but not sure it is the right way to handle this?

sethterashima, 2024-02-28:
The counter in question is included in each command, and should be incremented with each one. The car looks at the counter value to detect and reject replayed commands. If the car sees a replayed counter value, it returns MESSAGEFAULT_ERROR_INVALID_TOKEN_OR_COUNTER and sends the client an updated counter value to use.

Are you using this library or a different implementation? This library should recover from this error automatically. Some more logs would be helpful in diagnosing the problem (please redact your VIN from anything you post here).

If you're using your own protocol implementation, see [error recovery documentation](https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md#recovering-from-synchronization-errors).

robin-karlsson, 2024-02-29:
Yes, we're using the command proxy directly from this repository. This night it worked fine though. Looking through some of the other log samples, I noticed some epoch error as well. So it seems maybe more things were off.

Right now we run the proxy in two concurrent pods, maybe we should scale down to one to see if it works better?

robin-karlsson, 2024-02-29:
Later logs, including the response mentioned in the title.

```
{ "operationStatus": "OPERATIONSTATUS_ERROR", "signedMessageFault": "MESSAGEFAULT_ERROR_TIME_TO_LIVE_TOO_LONG" }

2024-02-28T00:16:19Z [debug] Executing charge_start on LRWYXYZ
2024-02-28T00:16:19Z [info ] Starting dispatcher service...
2024-02-28T00:16:19Z [info ] Session for DOMAIN_INFOTAINMENT loaded from cache
2024-02-28T00:16:19Z [info ] Session for DOMAIN_VEHICLE_SECURITY loaded from cache
2024-02-28T00:16:19Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/LRWYXYZ/signed_command: {"routable_message":"MgIIAzoSEhAhynxYI3Owx4cifGZ3ldajmgMQ86f+05eo+OLoDAeSy4cmZVIGEgQyAhIAaoMBCkMKQQQW58m5qYJ/T4hNUXvydZkICzYKvbKcJsiLDXkrLPf2bF6l4yB0FzXOxWfy0E1ghJ3ge/dSaeg+BP8dg/fX6c2NQjwKEFFmszUoclDEGCx0SDTCzYwQsAEdhrcBACIggMozyRyYPl/wOnKrfqF8gJcv4QnCxQhr8LsCh+CT6oM="}
2024-02-28T00:16:19Z [debug] Server returned 200: OK: {"response":"MhISECHKfFgjc7DHhyJ8ZneV1qM6AggDYgQIAhAUkgMQ86f+05eo+OLoDAeSy4cmZXpdCLIBEkEELM1Q5EENOCm5ELTOvkosKOEpaVqdm85RNNXBzFLBPChqMg7op7sNk2gODbDi543PP/UoY5sCWXEHX3MfrL8KoxoQUWazNShyUMQYLHRINMLNjCWf+wAAaiQyIgogyi7veYSRUUUlasncfHN8hifzIKwzsxJV1qLYJhu72lc="}
2024-02-28T00:16:19Z [info ] [f3a7fed397a8f8e2e80c0792cb872665] Updated session info for DOMAIN_INFOTAINMENT
2024-02-28T00:16:20Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/LRWYXYZ/signed_command: {"routable_message":"MgIIAzoSEhAhynxYI3Owx4cifGZ3ldajmgMQQQ2p1B5cs8HzG91Hgu5CylIGEgQyAhIAaoMBCkMKQQQW58m5qYJ/T4hNUXvydZkICzYKvbKcJsiLDXkrLPf2bF6l4yB0FzXOxWfy0E1ghJ3ge/dSaeg+BP8dg/fX6c2NQjwKEFFmszUoclDEGCx0SDTCzYwQswEdpfsAACIgmX8umyRAWGGy1L9avXhPG5grs/CI+9CcZaNYSjXUxyk="}
2024-02-28T00:16:20Z [debug] Server returned 200: OK: {"response":"MhISECHKfFgjc7DHhyJ8ZneV1qM6AggDYgCSAxBBDanUHlyzwfMb3UeC7kLKmgMQcDxMh6G4Q37QmrSLW2Ddm1IRCg8IARILCglyZXF1ZXN0ZWQ="}

{ "operationStatus": "OPERATIONSTATUS_ERROR", "signedMessageFault": "MESSAGEFAULT_ERROR_INVALID_TOKEN_OR_COUNTER" }

2024-02-28T03:16:19Z [info ] Received POST request for /api/1/vehicles/LRWYXYZ/command/charge_start
2024-02-28T03:16:19Z [debug] Executing charge_start on LRWYXYZ
2024-02-28T03:16:19Z [info ] Starting dispatcher service...
2024-02-28T03:16:19Z [info ] Session for DOMAIN_INFOTAINMENT loaded from cache
2024-02-28T03:16:19Z [info ] Session for DOMAIN_VEHICLE_SECURITY loaded from cache
2024-02-28T03:16:19Z [debug] Sending request to https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/vehicles/LRWYXYZ/signed_command: {"routable_message":"MgIIAzoSEhChasVPC+9h5Rd1zfu+BGL+mgMQ9boZB6AR1EWJ5Jp4AoSJaVIGEgQyAhIAaoMBCkMKQQQW58m5qYJ/T4hNUXvydZkICzYKvbKcJsiLDXkrLPf2bF6l4yB0FzXOxWfy0E1ghJ3ge/dSaeg+BP8dg/fX6c2NQjwKEFFmszUoclDEGCx0SDTCzYwQuAEduC8BACIgwIPYjpfejdkDI0/xayuSg41w1/4I+AVHd7HijR8uUR0="}
2024-02-28T03:16:20Z [debug] Server returned 200: OK: {"response":"MhISEKFqxU8L72HlF3XN+74EYv46AggDYgQIAhAGkgMQ9boZB6AR1EWJ5Jp4AoSJaXpdCLgBEkEELM1Q5EENOCm5ELTOvkosKOEpaVqdm85RNNXBzFLBPChqMg7op7sNk2gODbDi543PP/UoY5sCWXEHX3MfrL8KoxoQUWazNShyUMQYLHRINMLNjCUbCwEAaiQyIgogfFLEnckXqO8mMAwksEsd+Cih/tuyPOFpqP07ScsTjIs="}
2024-02-28T03:16:20Z [info ] [f5ba1907a011d44589e49a7802848969] Updated session info for DOMAIN_INFOTAINMENT
```

sethterashima, 2024-02-29:
MESSAGEFAULT_ERROR_INCORRECT_EPOCH is expected the first time you send a command to a car after it reboots (e.g., after an update). The car provides the correct epoch to use in the response and again this problem should correct itself.

> Right now we run the proxy in two concurrent pods, maybe we should scale down to one to see if it works better?

This could certainly explain the issue. If multiple clients are using the same private key to authorize commands but are not synchronizing the message counter, then the second client to send a command will get MESSAGEFAULT_ERROR_INVALID_TOKEN_OR_COUNTER. It will resync, and then the first client will get that error the next time it sends a command, etc.

To be clear, the clients should be retrying the commands after receiving that error and succeeding. But you'll introduce latency, hit API limits much faster, and have a higher timeout rate.

I don't know what your network topology is, but the simplest way to avoid these message counter conflicts while still using two pods would be to have each pod be responsible for a disjoint set of VINs.

If that's not possible, a more involved solution with N pods would be to modify the library so that Pod X only used counter values equal to X mod N; e.g., with N=4 pods, Pod 3 would use counter values 3, 7, 11, 15, ... . This would significantly reduce but not entirely eliminate the problem for INFOTAINMENT. The problem would remain for VEHICLE_SECURITY commands. Your logs should indicate how often you get "Updated session info for DOMAIN_INFOTAINEMNT" as opposed to "DOMAIN_VEHICLE_SECURITY".

closed 2024-03-01 by robin-karlsson