teslamotors/vehicle-command #200: Proxy: timeout error for single vehicle

andig, 2024-03-03:
We're running a proxy based on this module. Things are working fine, except for a single user that is suffering from timeout errors: https://github.com/evcc-io/evcc/issues/12600.

That is, sending any command over the proxy runs into timeout:

```
POST /api/1/vehicles/LRW.../command/set_charging_amps HTTP/1.1
{"charging_amps": 5}

ERROR 2024/03/03 10:39:49 max current: Post "https://.../api/1/vehicles/LRW.../command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
```

Vehicle is in EU region, our client timeout is 10s. When we increase the client timeout we receive this upstream response:

    {"response":null,"error":"context deadline exceeded","error_description":""}

Is this something that could be a backend issue for a specific vehicle?

andig, 2024-03-10:
User explained in https://github.com/evcc-io/evcc/issues/12600#issuecomment-1987307753 that deleting and re-installing the virtual key fixed the problem. Seems there might be an issue with half-installed virtual keys?

stx, 2024-03-11:
Yes, I've seen some instances of this as well. fleet_status also returns that the vehicle is unpaired even though the key is present, and it requires repairing.

I discovered an issue that the Tesla Android app doesn't like certain CDN caching headers on the hosted key and will either half-install the key or show "this party isn't registered with Tesla".

sethterashima, 2024-03-14:
This one appears to be either a backend issue, although some more investigation may be required. Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

In order to improve our response times, the scope of the Github issues tracker is being narrowed to the Golang code in the SDK. Backend issues should be reported to the Fleet API support team.