teslamotors/vehicle-command #203: Tesla app pairing instructions for non-owners

freddybushboy, 2024-03-04:
I’ve noticed an unintuitive experience for users who do not own their vehicle, somewhat related to [this issue](https://github.com/teslamotors/vehicle-command/issues/45). Essentially the Tesla app doesn't do the best job of setting expectations for what the user will need to do and the fact they will need to be in their vehicle to finish pairing.

1. When attempting to pair our user was shown the ownership error as expected
2. However the next screen doesn’t give the user any idea that they need to be in their vehicle. If they tap “approve”…
3. They see an error with no guidance for the user instructing them that they need to be in their vehicle
4. If they **are** in the car when they do this they do see some further instructions and are able to complete the pairing from there

|1|2|3|4|
---|---|---|---
|[image]|[image]|[image]|[image]

sethterashima, 2024-03-14:
Thanks for the feedback. I filed a ticket with the mobile app team.

GerardPolloRebozado, 2024-08-07:
dont know if its the same error but for me it says "this third person is not registered in tesla"

closed 2025-01-23 by thefireblade