teslamotors/vehicle-command #208: Incorrect Region

rawmean, 2024-03-08:
Initially I used the `region` endpoint to determine the Region and then I used the `ou_code` in the access token to determine the region. Both methods give the incorrect region for some users and I get the following error:

`user out of region, use base URL: https://fleet-api.prd.eu.vn.cloud.tesla.com, see https://developer.tesla.com/docs/fleet-api#regional-requirements`

itsMeDavidV, 2024-03-12:
So I have a hunch `refresh_token` prefixes are sometimes wrong as well. I have ID'd a few cases where the prefix pointed to the incorrect region for EU users. I cannot confirm, bc by design I have no way of inspecting a user's auth data, however error logs strongly suggest my theory.

sethterashima, 2024-03-14:
Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

In order to improve our response times, the scope of the Github issues tracker is being narrowed to the Golang code in the SDK. Backend issues should be reported to the Fleet API support team.

closed 2024-03-14 by sethterashima