teslamotors/vehicle-command #209: [Question] About register

Whirlwind, 2024-03-11:
I am reading the document and have some questions I would like to consult. Can someone help me?

1. The `register` (`/api/1/partner_accounts`) require the public key save at the `https://<app domain>/.well-known/appspecific/com.tesla.3p.public-key.pem`. This public key will be registered on the Vehicle, right? Then all vehicles will use a same public key, right? Then I use same private key and their VIN will be able to control any vehicle by BLE, right? 

2. If the user has already registered the public key through BLE, do I still need to call this register API?

3. I must call the `register` API (`/api/1/partner_accounts`) before vehicle owners open the link https://tesla.com/_ak/example.com. right?

sethterashima, 2024-03-11:
1. Yes, the public key will be paired with the vehicle when the owner opens https://tesla.com/_ak/example.com and approves. You'll use the private key and VIN to authorize commands. However, public keys that are paired this way cannot be used over BLE. See [documentation](https://github.com/teslamotors/vehicle-command/blob/main/pkg/protocol/protocol.md#roles). If you want to be able to use a command over BLE, you'll need to pair using [SendAddKeyRequest](https://pkg.go.dev/github.com/greenmission/vehicle-command/pkg/vehicle#Vehicle.SendAddKeyRequest) and use a previously paired NFC card.
2. Right.
3. If the key is paired over BLE, owners do not need to open the link.

closed 2024-03-11 by sethterashima

Whirlwind, 2024-03-12:
@sethterashima Thank you for your reply. as you said, I have some questions:

1. `public keys that are paired this way cannot be used over BLE`, so the `Fleet Manager` be not able to drive the car, (only send the command to vehicle by remote api), right?
2. If My app will only support the Fleet API, is there a key be shown in the vehicle screen like a bluetooth key?
3. If My app will support both the Fleet API to remote control the vehicle and the BLE to local control the vehicle, I need two public keys to paired, right?

Whirlwind, 2024-03-12:
> You'll use the private key and VIN to authorize commands
So, If I get the private key from other app, I will remote control all vehicles which use the app, right? This seems very unsafe.