teslamotors/vehicle-command #214: Unable to share vehicle

NikhilTank35, 2024-03-13:
When the vehicle owner scans https://www.tesla.com/_ak/app.stage.emulate.network, it gives "Unable to share vehicle, this third party is not registered with Tesla......" error but https://www.tesla.com/_ak/blinded-standard.000webhostapp.com works which have same public key present.

[image] 

[image]

NikhilTank35, 2024-03-13:
@ThomasAlxDmy @sethterashima @fabianhu Can you please help us with this?

NikhilTank35, 2024-03-13:
Is it like only the `*.com` domain works ?

sethterashima, 2024-03-14:
Please see https://developer.tesla.com/docs/fleet-api#help-and-support.

In order to improve our response times, the scope of the Github issues tracker is being narrowed to the Golang code in the SDK. Backend issues should be reported to the Fleet API support team.

closed 2024-03-14 by sethterashima